MicrosoftDocs/windows-driver-docs-ddi #1237: Is normal that HTREE\ROOT\0 is 'unknow'

RobertoB74, 2022-01-02:
i am beginner in a drivers devellopment .... if i have understand correctly this is the root of the hierarchy ...
but is it not weird that is 'Unkow' the class, producer and so on?

istance ID:  HTREE\ROOT\0
description: Unknow
class name:  Unknow
GUID class:  Unknow
producer:    Unknow
State:       Start

tedhudek, 2022-01-05:
Hi @RobertoB74, is there a specific doc page you are asking about?

If you are looking for general technical assistance, I suggest posting your question in the following forums, where driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [windows-hardware-wdk](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk.html)
* [OSR's NTDEV forum](https://community.osr.com/categories/ntdev)

tedhudek, 2022-01-07:
Closing out due to lack of info, please reopen if still needed

closed 2022-01-07 by tedhudek