MicrosoftDocs/windows-driver-docs-ddi #1240: question about film_grain.ar_coeffs_y

tong1wu, 2022-01-06:
In the document as well as DXVA_AV1 spec, for parameter film_grain.ar_coeffs_y, it says:
`The auto-regressive coefficients for the Y plane. This field corresponds to the ar_coeffs_y_plus_128 syntax element from the AV1 specification.`

I wonder if the 'corresponds to' really literally means ar_coeffs_y = ar_coeffs_y_plus_128 or ar_coeffs_y equals to ar_coeffs_y_plus_128 -128. 

Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 346c31e9-85e3-c040-d105-5a86aa3ceffe
* Version Independent ID: c1f89a9b-d502-d0c8-17bb-db94480964eb
* Content: [DXVA_PicParams_AV1 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/dxva/ns-dxva-dxva_picparams_av1#feedback)
* Content Source: [wdk-ddi-src/content/dxva/ns-dxva-dxva_picparams_av1.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dxva/ns-dxva-dxva_picparams_av1.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

mattrwoz, 2022-01-25:
film_grain.ar_coeffs_y is identical to the AV1 specification value ar_coeffs_y_plus_128.  So, if you have ar_coeffs_y without the 128 offset then you need to add it back on: film_grain.ar_coeffs_y = ar_coeffs_y + 128.  The same is true for ar_coeffs_cb and ar_coeffs_cr.

tong1wu, 2022-01-25:
Thanks, it helps. Closing this issue.

closed 2022-01-25 by tong1wu