MicrosoftDocs/windows-driver-docs-ddi #1241: portcls.h incorrectly defines IMP_IMiniportWaveRTOutputStream macro

emuzychenko, 2022-01-07:
In all 10.x WDKs, portcls.h defines IMP_IMiniportWaveRTOutputStream macro with STDMETHOD_ instead of STDMETHODIMP_.

It causes warning C4986 (exception specification does not match previous declaration). This warning is off by default, /Wall turns it on.

DOMARS, 2022-02-10:
@emuzychenko - I wonder if you have an additional include, or something else is going on in the compile settings, that is causing the "C4986 (exception specification does not match previous declaration)." 

https://docs.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-c4986?view=msvc-170

*This warning can be generated when there is an exception specification in one declaration and not the other.*

As I see a mix of STDMETHOD_ and STDMETHODIMP_. For example, part of the define for IMiniportWavePci.

```cpp
    //  For IMiniportWavePci
    STDMETHOD_(NTSTATUS,Init)
    (   THIS_
        _In_      PUNKNOWN            UnknownAdapter,
        _In_      PRESOURCELIST       ResourceList,
        _In_      PPORTWAVEPCI        Port,
        _Out_     PSERVICEGROUP *     ServiceGroup
    )   
```

My belief is that these macros have been this way for a while, so perhaps something else in your WDK build environment is causing issues, as the PortCls header is in widespread use.

The Raymond Chen blog series describes the use of these macros.
https://devblogs.microsoft.com/oldnewthing/20200911-00/?p=104205

For additional questions on the use of WDK and compile issues, I would recommend the OSR dev group. [OSR's ntdev List](https://community.osr.com/categories/ntdev)

Good luck with your use of portcls, and WDM audio kernel driver development.

closed 2022-02-10 by DOMARS

emuzychenko, 2022-02-11:
I'm afraid you misunderstood me. I mean that all IMP_XXX macros except IMP_IMiniportWaveRTOutputStream use STDMETHODIMP_, and only IMP_IMiniportWaveRTOutputStream uses STDMETHOD_.