MicrosoftDocs/windows-driver-docs-ddi #1243: Needs flag value

poweradminllc, 2022-01-12:
Would be great to have flag values, such as POOL_NX_ALLOCATION.

Thanks for otherwise great docs!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e596ca62-205d-9b48-3f41-691977044a86
* Version Independent ID: 2cc29109-02e5-ac7f-4055-7cbdd08ce881
* Content: [ExInitializeNPagedLookasideList function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exinitializenpagedlookasidelist?f1url=%3FappId%3DDev15IDEF1%26l%3DEN-US%26k%3Dk(FILESIGHTMF%252FExInitializeNPagedLookasideList);k(ExInitializeNPagedLookasideList);k(DevLang-C%252B%252B);k(TargetOS-Windows)%26rd%3Dtrue)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exinitializenpagedlookasidelist.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exinitializenpagedlookasidelist.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-01-13:
Hi @poweradminllc, thanks for the kind words. For your request, do you mean the numerical values for those flags, i.e. (from wdm.h):

```
#define POOL_NX_ALLOCATION   512
#define POOL_RAISE_IF_ALLOCATION_FAILURE 16
```

poweradminllc, 2022-01-13:
> Email: ***@***.***
>  Web: [](https://www.poweradmin.com/?ref=helpspot) [https://www.poweradmin.com/](https://www.poweradmin.com/)
>  Support: [](http://support.poweradmin.com/?ref=helpspot) [http://support.poweradmin.com/](http://support.poweradmin.com/)
>  Twitter: ***@***.***(http://twitter.com/poweradmn)
>  Phone: +1 800.401.2339
>  Fax: +1 866.266.8330
>   If you were delighted by our service, help us get the word out!
>   Send a tweet, post to a forum, or just tell your buddy.
>   We would really appreciate it.  Thanks :)
> Ted Hudek - ***@***.*** (Jan 12 2022, 06:10 PM)
> -----------------------------
>
> Hi @poweradminllc, thanks for the kind words. For your request, do you mean the numerical values for those flags, i.e. (from wdm.h):
>
> #define POOL_NX_ALLOCATION   512
> #define POOL_RAISE_IF_ALLOCATION_FAILURE 16

## Reply ABOVE THIS LINE to add a note to this request ##
-------------------------------------------

Yes, having the numerical values in the documentation would be a huge help so we don't have to go hunt them down.

Doug  
*Power Admin LLC*

poweradminllc, 2022-01-13:
> Ted Hudek - ***@***.*** (Jan 13 2022, 11:33 AM)
> Hi Doug, I&#039;m not a driver dev myself so please forgive the question, but doesn&#039;t the pound define equate the flag name with the value, so that you can use the name interchangeably in the code?
> I&#039;m curious to learn the scenario where you need the value, because there are tons of such flags in our docs.

## Reply ABOVE THIS LINE to add a note to this request ##
-------------------------------------------

Hi Ted -

You are right - the #define does equate to the value.   There are two cases where knowing the #define doesn't help you though:

1. You're using the value with headers that don't include the value (for example when compiling code for both Windows 8 and Windows 10, and the value is defined for Windows 10, but the Windows 8 headers are used).

I have such a project where I have this line of code:

ULONG initLALFlags = 512; //POOL_NX_ALLOCATION;

since POOL_NX_ALLOCATION isn't defined in all projects/build situations where that code is used.

2. When debugging and looking at a value in code (i.e. a value received from the OS for example) - the #define doesn't help at all in that case - you only have the raw value to compare against.  Another situation might be where I'm looking at values I set in my code (myVal = POOL_NX_ALLOCATION) and I can see that the variable is set to 512, but I'd like to be able to verify it has the correct documented value.  These situations come up a lot.

Thank you for asking.

 

Doug  
*Power Admin LLC*

tedhudek, 2022-01-14:
Hi Doug,

Thanks for the explanation, that's helpful.

Were you thinking just for this page, or is there a small set of pages you use regularly that you think would benefit from having the values added?

poweradminllc, 2022-01-14:
> Ted Hudek - ***@***.*** (Jan 13 2022, 07:29 PM)

## Reply ABOVE THIS LINE to add a note to this request ##
-------------------------------------------

Just this page, but in general, it would be very helpful to have it on all pages.   Naturally that won't get done any time soon, but if it could be done anytime a page is updated... ?

 

Doug  
*Power Admin LLC*

tedhudek, 2022-01-14:
Ok Doug, the numerical values now appear in the description of Flags on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exinitializenpagedlookasidelist.

I've also discussed it with the team and folks agreed that it's good to add the numerical values for constants going forward.  

Thanks for raising this!

closed 2022-01-14 by tedhudek