MicrosoftDocs/windows-driver-docs-ddi #1244: FltQueueDeferredIoWorkItem for IRP_MJ_WRITE, IRP_MJ_READ, and IRP_MJ_FLUSH_BUFFERS

scottnoone, 2022-01-19:
For the life of me I cannot figure out what this means or what case it is referring to:

**Caution   To avoid deadlocks, a minifilter must not post an I/O operation to a system work queue in the post-operation callback for any I/O operations that a driver can complete directly in the storage stack, such as the following:
IRP_MJ_READ
IRP_MJ_WRITE
IRP_MJ_FLUSH_BUFFERS**

Can this be clarified? I don't see any reason why you couldn't post a post-Write/Read/Flush arriving at DISPATCH_LEVEL. The fact that you're at DISPATCH_LEVEL means the underlying IRP is probably being completed in a DPC, which would mean that the operations was posted along the way anyway. 

Note that FltDoCompletionProcessingWhenSafe docs also say this and that's where I get the most questions about this. Also, the SwapBuffers sample calls FltDoCompletionProcessingWhenSafe in its post-write.

Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0c68fdef-072f-0a16-60d5-c0c3c2b04c93
* Version Independent ID: 5307944c-6802-a6bb-c58c-239eb0d94bdc
* Content: [FltQueueDeferredIoWorkItem function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltqueuedeferredioworkitem)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltqueuedeferredioworkitem.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltqueuedeferredioworkitem.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2022-01-19:
Agree with @scottnoone 

There are many many things which get in the way during post read and write and I have been bitten by most of them. 

I'll also observe that I would never use this function whilst its semantics are so loosely defined.  For instance:
- If you get a failure back, who owns the CBD?  
- Is this true for all failure statuses or some?  
- What is the width of the queue that you are posting to (if you post an MPW write to a queue and the queue is held up by a pagefault then you risk deadlocking the machine). 
- What priority does the thread run at (are you going to get priority inversion?).

Any or all of these are reasons to treat this function with respect in the cases mentioned.    Mostly I have interpreted it as a warning against the FSD locks that are held by the file system in pre.  Me?  I just implement my own thread queue and move on.

lorihollasch, 2022-03-05:
Thanks for filing this issue, @scottnoone, and thanks @rodwiddowson for sharing your observations. I was finally able to meet with a few sage devs about the Caution note. It turns out that the condition for which it was added no longer exists, and the Caution text is now indeed wrong. I've removed it from FltQueueDeferredIoWorkItem and FltDoCompletionProcessingWhenSafe.

closed 2022-03-05 by lorihollasch

jxy-s, 2022-08-25:
> It turns out that the condition for which it was added no longer exists.

Could we instead be provided with information on _when_ this changed? A clarification would be preferred over removing the caution entirely. Someone down the line may stumble into an unfortunate bug in previous versions of the OS.

lorihollasch, 2022-08-25:
@jxy-s, it was removed back in the days of Windows 2000 - long enough ago that we don't (can't) track that level of detail.