MicrosoftDocs/windows-driver-docs-ddi #1246: Where IRP_MN_QUERY_DEVICE_RELATIONS request in WDF?

Itsaniceday, 2022-01-24:
How could I get the IRP_MN_QUERY_DEVICE_RELATIONS request of the information or some members‘ in IRP structure？Such as, Irp->IoStatus.Information in WDM module.I tried to call the WdfRequestGetInformation function,but I didn't know how could I get the "request".Could you provide some samples?
I am looking forward to hearing from you soon. 
Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da18b960-903b-484f-cd7b-d47e79a246c4
* Version Independent ID: 14be5e02-0a16-dfd3-108d-2e2cc8dcb0aa
* Content: [EVT_WDF_DEVICE_RELATIONS_QUERY (wdfdevice.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfdevice/nc-wdfdevice-evt_wdf_device_relations_query)
* Content Source: [wdk-ddi-src/content/wdfdevice/nc-wdfdevice-evt_wdf_device_relations_query.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfdevice/nc-wdfdevice-evt_wdf_device_relations_query.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-26:
Hi @Itsaniceday, thanks for the question.  I might be able to help you better if you can describe what specifically is it that you would like to do that cannot be done with the existing EvtWdfDeviceRelationsQuery callback?

Itsaniceday, 2022-01-27:
Hi @tedhudek, It was can be run when the driver received an IRP_MN_QUERY_DEVICE_RELATIONS request. I just don't understand how could I get the WDFREQUEST for IRP_MN_QUERY_DEVICE_RELATIONS this request. I want to get this request,but the parameters of **WDFREQUEST request** not existing with **EvtWdfDeviceRelationsQuery( [in] WDFDEVICE Device,  [in] DEVICE_RELATION_TYPE RelationType)** callback.

tedhudek, 2022-01-27:
Hi @Itsaniceday,

Thanks, and understood, but my question remains -- why do you need the request?  I ask because one goal of WDF is to make it possible for driver writers to avoid dealing with WDM and handling IRPs directly.  So, if there is a scenario where WDF does not support this, it would be helpful if we could identify it.

BTW, in looking at the page in question (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfdevice/nc-wdfdevice-evt_wdf_device_relations_query), I noticed that descriptions for RelationType were missing from the Remarks section.  I just pushed a fix for this - those should appear on the live site within 30 min.

Itsaniceday, 2022-01-28:
Hi @tedhudek, it's helpful, and thanks your proposal.

Latterly I wanna do a porting work from wdm to kmdf. It has had to be got or set this IRP(request). If the request couldn't be got that I haven't any ideas to process it. Is existing any way to get **Irp->IoStatus.Information** for the kmdf module? If you can provide some examples that will be understood as soon.

tedhudek, 2022-01-28:
Hi @Itsaniceday,

Are you writing a bus driver, and you want to allocate PDEVICE_RELATIONS and assign it to Irp->IoStatus.Information?
  
If not, which kind of DEVICE_RELATION_TYPE is your existing WDM driver dealing with?

You can find a list of DEVICE_RELATION_TYPE values on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ioinvalidatedevicerelations.

Itsaniceday, 2022-01-29:
Hi @tedhudek, 

#Are you writing a bus driver, and you want to allocate PDEVICE_RELATIONS and assign it to Irp->IoStatus.Information?
Yeah,that is it. Is existing some functions dealing with it?

tedhudek, 2022-01-31:
Good morning @Itsaniceday,

Great!  You should be able to find the information you need by starting at https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/enumerating-the-devices-on-a-bus.

If you don't find what you need there, feel free to re-open this issue.  Best of luck!

closed 2022-01-31 by tedhudek