MicrosoftDocs/windows-driver-docs-ddi #1247: struct TIME_FIELDS should be struct _TIME_FIELDS

johannesthoma, 2022-01-24:
On: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtltimefieldstotime
the TIME_FIELDS struct is documented. However the struct is really called struct _TIME_FIELDS (like all other kernel structures with a leading _ ). not struct TIME_FIELDS.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f8ed35c1-00c4-8cca-7974-fa703436886e
* Version Independent ID: d8f8fa38-08b7-9e80-ea27-551b3a26fbba
* Content: [RtlTimeFieldsToTime function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtltimefieldstotime)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtltimefieldstotime.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtltimefieldstotime.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

johannesthoma, 2022-01-24:
Best regards, Johannes

tedhudek, 2022-01-26:
Thanks for the report Johannes!  I just pushed 1806ccb352bd9eead6d6ca4d71d4598646e664f8, which splits out that struct into a new page of its own.  I realize that we still refer to it without the underscore -- that's just convention in this particular doc set.

The new rendered page should arrive on docs.microsoft.com within 30 minutes or so -- expected URL will be https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-time_fields.

Thanks again for caring about the docs :)

closed 2022-01-26 by tedhudek