MicrosoftDocs/windows-driver-docs-ddi #1249: What happens if WdfTimerStop() is called twice on the same timer object

prakhya, 2022-01-28:
Could you please explain in the document what happens if WdfTimerStop() is called twice on the same timer object?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 423f563c-6557-cc53-84de-2347b5a0587f
* Version Independent ID: 375f88d1-32e7-4b9d-4944-dc4382d1ad47
* Content: [WdfTimerStop function (wdftimer.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdftimer/nf-wdftimer-wdftimerstop)
* Content Source: [wdk-ddi-src/content/wdftimer/nf-wdftimer-wdftimerstop.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdftimer/nf-wdftimer-wdftimerstop.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2022-01-28:
Hi @prakhya and thanks for the question.  We need some time to test this out.  I expect to reply further next week.

prakhya, 2022-01-31:
Hi @tedhudek, thanks for the reply.

tedhudek, 2022-02-05:
Sorry @prakhya looks like we are going to need another week on this one.  We haven't forgotten!

prakhya, 2022-02-08:
Sure! no worries! Thanks for letting me know.

tedhudek, 2022-02-16:
Hi @prakhya,

There are several variations: 
 1) whether WdfTimerStop is called with Wait=TRUE or not
 2) whether the two Stop calls use the same Wait value or not
 3) whether the two Stop calls are from different threads or not
 4) whether the timer was created as a periodic one or not
 5) whether WdfTimerStart is called from another thread or not

Code is here:

https://github.com/microsoft/Windows-Driver-Frameworks/blob/bfee6134f30f92a90dbf96e98d54582ecb993996/src/framework/shared/core/fxtimer.cpp#L653

Assume Start is not called between the two Stop calls. Looking at the code, the second call is always an empty operation. There is a chance that WDF will assert (when verifier is on) and continue if the driver calls Stop(Wait=TRUE) from different threads (line 709).

Hope that helps.  Let me know if you feel any doc changes are needed to clarify.

closed 2022-02-16 by tedhudek