MicrosoftDocs/windows-driver-docs-ddi #1250: RegNtPreSaveMergedKey and RegNtPostSaveMergedKey information is not documented

camptatopenmars, 2022-01-29:
RegNtPreSaveMergedKey and RegNtPostSaveMergedKey information is not documented.  These appear to be new registry callbacks with windows 10 (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_reg_notify_class)  However their associated information class is not documented (though is included in the wdm.h as REG_SAVE_MERGED_KEY_INFORMATION).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b00fb9cf-79c4-d882-77a2-7ffbf8252a1d
* Version Independent ID: 7ac69fad-549e-0e28-ccb3-906c44736edf
* Content: [EX_CALLBACK_FUNCTION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nc-wdm-ex_callback_function)
* Content Source: [wdk-ddi-src/content/wdm/nc-wdm-ex_callback_function.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nc-wdm-ex_callback_function.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-01-31:
Hi @camptatopenmars are you requesting that we add member descriptions to the existing doc page for the struct [REG_SAVE_MERGED_KEY_INFORMATION](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-reg_save_merged_key_information)?

camptatopenmars, 2022-01-31:
REG_SAVE_MERGED_KEY_INFORMATION seems to be missing from the list at
EX_CALLBACK_FUNCTION
(wdm.h) - Windows drivers | Microsoft Docs
<https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nc-wdm-ex_callback_function>
which
is all I was reporting here.

However... yes adding member descriptions to REG_SAVE_MERGED_KEY_INFORMATION
<https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-reg_save_merged_key_information>
would
go a long way towards providing value beyond the header declaration by
documenting what the members mean.  High/Low and what exactly is merged
between them and the object key in particular.

t.

t.

tedhudek, 2022-02-01:
Update -- I added the values to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nc-wdm-ex_callback_function but am still looking for info about https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-reg_save_merged_key_information.

tedhudek, 2022-02-05:
Update - https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-reg_save_merged_key_information is now filled out.

Thanks again for the request - we appreciate your interest in the docs :)

closed 2022-02-05 by tedhudek

camptatopenmars, 2022-02-05:
Thanks for making the docs!

t.