MicrosoftDocs/windows-driver-docs-ddi #1259: Somewhat confusing parameter description.

camptatopenmars, 2022-02-07:
The description "Provides the address of the notification information for the specified type of thread notification." is confusing since 'information' is typically thought of as data - such as C struct.  I believe this is actually the address of the _callback_ for the requested thread notification information.  This is particularly confusing since the function is prototyped oddly in the WDK to not clearly indicate a callback function or its prototype.  We can of course infer that it is probably the same prototype as used for the non-'Ex' variant of this API and https://github.com/OSRDrivers/kmexts/blob/master/kmexts.cpp demonstrates that is a correct inference. 

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d768aaa3-7a6a-27aa-ee25-a4a0888b3f21
* Version Independent ID: a27fab1b-d0a1-0502-9b2a-c91c701bdddc
* Content: [PsSetCreateThreadNotifyRoutineEx function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-pssetcreatethreadnotifyroutineex)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-pssetcreatethreadnotifyroutineex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-pssetcreatethreadnotifyroutineex.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-02-09:
Great point @camptatopenmars.  For the current two values of https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ne-ntddk-_pscreatethreadnotifytype, the NotifyInformation parameter of 
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-pssetcreatethreadnotifyroutineex should be of type https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nc-ntddk-pcreate_thread_notify_routine, you are correct.

I've added an extra sentence to the description of the NotifyInformation parameter to clarify that.  That change should appear on the live site within half an hour.

Thanks for the feedback!

closed 2022-02-09 by tedhudek

camptatopenmars, 2022-02-09:
thank you again!

t.