MicrosoftDocs/windows-driver-docs-ddi #1260: Return type for MmGetMdlVirtualAddress

prakhya, 2022-02-08:
The function definition says return type is void (void MmGetMdlVirtualAddress) but the remark section says that the function returns starting virtual address of the MDL. Could you please clarify what does the function return?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 08454e64-36f8-2a12-acfe-dbaa8279f7de
* Version Independent ID: ba6bcd6b-fa1f-51af-2463-a9464edd2791
* Content: [MmGetMdlVirtualAddress - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-mmgetmdlvirtualaddress)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-mmgetmdlvirtualaddress.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-mmgetmdlvirtualaddress.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-02-09:
Hey Sai - thx for the feedback.  Looks like it's definitely not retval void.  See https://github.com/microsoft/Windows-driver-samples/search?q=MmGetMdlVirtualAddress.

From wdm.h:

```
//++
//
// PVOID
// MmGetMdlVirtualAddress (
//     _In_ PMDL Mdl
//     )
//
// Routine Description:
//
//     The MmGetMdlVirtualAddress returns the virtual address of the buffer
//     described by the Mdl.
//
// Arguments:
//
//     Mdl - Pointer to an MDL.
//
// Return Value:
//
//     Returns the virtual address of the buffer described by the Mdl
//
//--

#define MmGetMdlVirtualAddress(Mdl)                                     \
    ((PVOID) ((PCHAR) ((Mdl)->StartVa) + (Mdl)->ByteOffset))
```

Will work on updating the doc.

tedhudek, 2022-02-09:
Update - doc page now has the correct syntax block -- https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-mmgetmdlvirtualaddress

Thanks again for bringing this to our attention @prakhya!

closed 2022-02-09 by tedhudek

prakhya, 2022-02-09:
@tedhudek Thanks for updating the doc.