MicrosoftDocs/windows-driver-docs-ddi #1266: Confusing warning in PLOAD_IMAGE_NOTIFY_ROUTINE documentation

scottnoone, 2022-02-15:
_"Warning  The actions that you can perform in this routine are restricted for safe calls. See Best Practices"_

Two things:

1. What is, "are restricted for safe calls" supposed to mean?
2. The Best Practices link is clearly incorrect. It's possible that a corrected link would answer #1, though I can't find anything on MSDN defining or describing "safe  calls".

Thanks!

-scott

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a9d0bfa8-68c2-0594-eaf3-8a77d4404157
* Version Independent ID: c8e2634a-a965-fb18-e68f-07b6682605ae
* Content: [PLOAD_IMAGE_NOTIFY_ROUTINE (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nc-ntddk-pload_image_notify_routine)
* Content Source: [wdk-ddi-src/content/ntddk/nc-ntddk-pload_image_notify_routine.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nc-ntddk-pload_image_notify_routine.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-02-17:
Hi @scottnoone thanks for the feedback.  I found the problem with the Best Practices link.  Will you let me know if that link clarifies at all?  If not happy to try and get more info on it.

tedhudek, 2022-02-22:
Closing out for now, please re-open if you'd like to request doc updates

closed 2022-02-22 by tedhudek