MicrosoftDocs/windows-driver-docs-ddi #1269: Return NTSTATUS code list not complete.

MartinCHarvey, 2022-02-22:
IoAcquireRemoveLock can also return 0xc00000bb STATUS_NOT_SUPPORTED in some unusual cases.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93b50201-ea38-2862-1ae4-b72dddf23d4c
* Version Independent ID: 72388b48-638a-6729-87cf-a35b2bdb303c
* Content: [IoAcquireRemoveLock macro (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ioacquireremovelock)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-ioacquireremovelock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ioacquireremovelock.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-03-08:
Hi MartinCHarvey, thanks for the feedback.  I am not seeing STATUS_NOT_SUPPORTED in the code.

What was the situation in which you received this return value?

MartinCHarvey, 2022-03-15:
Hi Ted,

I’m not surprised. Some of the IRP’s filtering through the driver produced this return code – but, they were of unknown type, and it was in the middle of a busy boot, so I’m suspecting that it’s either a transient start-up condition in a lower level driver, or a race with some entry points being set up. I suppose if it’s not an “official” return code, then you probably don’t have to document it.

MH.

tedhudek, 2022-03-16:
Ok Martin, appreciate the context.  That sounds entirely plausible to me.  I'll leave it off the page for now then.  Thanks for reporting it!

closed 2022-03-16 by tedhudek