MicrosoftDocs/windows-driver-docs-ddi #1271: Are we required to clone an NBL before calling FwpsInjectMacReceiveAsync?

henrikalmeida, 2022-02-28:
In the documentation, it is said that:

`A pointer to a [NET_BUFFER_LIST](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/nbl/ns-nbl-net_buffer_list) structure that describes the packet data that is being injected. A callout driver allocates a NET_BUFFER_LIST structure to use to inject packet data by calling either the [FwpsAllocateCloneNetBufferList0](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpsallocateclonenetbufferlist0) function or the [FwpsAllocateNetBufferAndNetBufferList0](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpsallocatenetbufferandnetbufferlist0) function.`

But are we required to clone an NBL before calling this?
Lets say we have referenced an NBL, and out of band we are trying to reinject it (without modifying it), are we allowed to use the original NBL when injecting using FwpsInjectMacSendAsync and FwpsInjectMacReceiveAsyn, and then dereferencing it in the completion?
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c56bfbf7-6db8-821a-5ab0-c56134f6b46b
* Version Independent ID: 299c0756-dbef-5eb4-e4e0-92feb4a64656
* Content: [FwpsInjectMacReceiveAsync0 function (fwpsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpsinjectmacreceiveasync0)
* Content Source: [wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjectmacreceiveasync0.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjectmacreceiveasync0.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2022-03-18:
Hi @henrikalmeida, Thanks for opening this issue. I passed along your question to the feature team but haven't received a response. In the meantime, you may try posting a question on [Microsoft Q&A](https://docs.microsoft.com/en-us/answers/products/) or the [OSR NTDEV](https://community.osr.com/categories/ntdev) forum.

aviviano, 2022-04-01:
Hi @henrikalmeida. I have not gotten a response from the feature team on your question so I am going to close this issue. If you are still having issues, you can try the [OSR NTDEV](https://community.osr.com/categories/ntdev) forum.

closed 2022-04-01 by aviviano