MicrosoftDocs/windows-driver-docs-ddi #1285: PhysicalAddress should represent GPU address before WDDM 2.0?

chaotsungpeng, 2022-04-07:
it seems m_dstGpuAddress & m_srcGpuAddress is the GPU address.
[m_bSwCommandBuffer](https://github.com/microsoft/graphics-driver-samples/blob/master/render-only-sample/roskmd/RosKmdRapAdapter.cpp#L358)

but above addresses are patched below with allocation->PhysicalAddress
https://github.com/microsoft/graphics-driver-samples/blob/master/render-only-sample/roskmd/RosKmdAdapter.cpp#L1603

thus it's not likely that allocation->PhysicalAddress is the physical address, but it's more like the gpu address, right?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 69682ac4-a065-8b83-8f16-086463be977d
* Version Independent ID: 9e0069d3-73b5-939f-3935-245f8c68e87b
* Content: [_DXGK_ALLOCATIONLIST (d3dkmddi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmddi/ns-d3dkmddi-_dxgk_allocationlist)
* Content Source: [wdk-ddi-src/content/d3dkmddi/ns-d3dkmddi-_dxgk_allocationlist.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmddi/ns-d3dkmddi-_dxgk_allocationlist.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2022-04-27:
@chaotsungpeng, the sample driver that you are referencing appears to be based on WDDM 1.3, which was before GPU virtual memory was introduced. The dev team indicated that you can think of the address that you're referencing to be a "GPU physical address". The description of PhysicalAddress on this refpage is correct, and is for Windows 10 (WDDM 2.0) and later.

closed 2022-04-27 by lorihollasch

chaotsungpeng, 2022-04-27:
@lorihollasch 
Thank you very much for your kind reply, but can I ask other questions regarding this issue?
1. The dev team indicated it's "GPU physical address", but it's not the real physical address, right? So can I say it's "GPU segment address" instead?
2. regarding WDDM 2.0, I have searched hard to find a sample code but in vain, can the dev team give me some suggestion about where to find the sample code? 
Best Regards

lorihollasch, 2022-04-27:
@chaotsungpeng:
You can find memory-related documentation in the following places:

- [Handling Memory Segments](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/handling-memory-segments) was written for original WDDM 1.0 to describe what a driver should do regarding memory segments
- [GPU Virtual Memory](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/gpu-virtual-memory-in-wddm-2-0) references changes relevant for WDDM 2.0

I'm unaware of a WDDM 2.0 sample, sorry. The Compute Only Sample in the repo you referenced might help? Otherwise, since this is a docs repo, you can try asking your questions on [Windows Hardware Development Q&A](https://docs.microsoft.com/en-us/answers/topics/windows-hardware-wdk-general.html) or the [OSR NTDEV forum](https://community.osr.com/categories/ntdev).