MicrosoftDocs/windows-driver-docs-ddi #1296: HIDP_CAPS and InputReportByteLength/OutputReportByteLength/FeatureReportByteLength

DJm00n, 2022-04-21:
Seems report ID size (1 byte) is always prepended to the report data and InputReportByteLength/OutputReportByteLength/FeatureReportByteLength - even if report IDs are not used in HID Report Descriptor.

Please confirm this and fix the page accordingly.

```
InputReportByteLength

Specifies the maximum size, in bytes, of all the input reports (including the report ID, if report IDs are used, which is prepended to the report data).

OutputReportByteLength

Specifies the maximum size, in bytes, of all the output reports (including the report ID, if report IDs are used, which is prepended to the report data).

FeatureReportByteLength

Specifies the maximum length, in bytes, of all the feature reports (including the report ID, if report IDs are used, which is prepended to the report data).
```

Same applies to `HidD_GetInputReport`/`HidD_GetFeature`/`HidD_SetFeature`/`HidD_SetOutputReport`'s `ReportBufferLength`.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c659f9ab-41d9-1898-439e-7d4eae1a3cab
* Version Independent ID: 5f131874-7fa6-f239-d726-3facd3f36dda
* Content: [_HIDP_CAPS (hidpi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/hidpi/ns-hidpi-_hidp_caps)
* Content Source: [wdk-ddi-src/content/hidpi/ns-hidpi-_hidp_caps.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/hidpi/ns-hidpi-_hidp_caps.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2022-04-26:
Hi @DJm00n

Thank you for this very thorough report. I am researching this and will update the topics when I have more information.

Thanks!
 ~ *Mark*

mhopkins-msft, 2022-04-28:
Hi @DJm00n,

To me, this is saying that the ID byte is always present. If it is not being used, the value of the ID byte will be zero. Therefore, I don't think this is incorrect, perhaps it could just be better written.

Is that consistent with your experience?

Thanks,
 ~ *Mark*

DJm00n, 2022-04-28:
Hi @mhopkins-msft!
Yes, that is consistent. Current wording is somewhat confusing and misleading. 

Adding @samtertzakian and @nefarius. Maybe they have some thoughts on this topic.

mhopkins-msft, 2022-04-29:
Hi @DJm00n,

Thank you for verifying. I've updated the topics to hopefully make them clearer. They should be republished soon.

As always, thank you for helping make our docs better!

 ~ *Mark*

closed 2022-04-29 by mhopkins-msft

DJm00n, 2022-04-29:
Thank you @mhopkins-msft!

One more thing.

Can you please add something like `This size is specified in InputReportByteLength/OutputReportByteLength/FeatureReportByteLength member of a HIDP_CAPS structure returned from HidP_GetCaps call.` to the to `HidD_GetInputReport`/`HidD_GetFeature`/`HidD_SetFeature`/`HidD_SetOutputReport`'s `ReportBufferLength` decription.