MicrosoftDocs/windows-driver-docs-ddi #1308: IrqPolicyAllProcessorsInMachineWhenSteered description missing

msudhakargit, 2022-05-20:
[Enter feedback here]
Hi
Please add description for "IrqPolicyAllProcessorsInMachineWhenSteered" and how is this differeent from "IrqPolicyAllProcessorsInMachine" and any guidance on when to choose which.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 17aeaf56-8a5f-dd74-32f1-d8d80485e22e
* Version Independent ID: 3bab9ec5-1bdc-32a1-34d9-809e66110599
* Content: [_IRQ_DEVICE_POLICY (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_irq_device_policy)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_irq_device_policy.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_irq_device_policy.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

mhopkins-msft, 2022-08-18:
Hi @msudhakargit,

Thank you for pointing this out to us. I've added a description for the **IrqPolicyAllProcessorsInMachineWhenSteered** value. The update should be published later today.

For more information about PCI bus IRQ steering, check out [this article](https://www.e-articles.info/e/a/title/What-is-PCI-Bus-IRQ-Steering-and-how-does-it-work/).

Thanks!
 ~ *Mark*

closed 2022-08-18 by mhopkins-msft

msudhakargit, 2022-08-19:
Thanks Mark!

mhopkins-msft, 2022-08-19:
@msudhakargit, thank you for helping make the docs better!