MicrosoftDocs/windows-driver-docs-ddi #1334: Incorrect and incomplete enum documentation

levicki, 2022-07-15:
This documentation is totally useless and incorrect.

It leads you to believe that values in the `enum` start from zero, are sequential, and never provides actual enumeration values.

Actual definition taken from `d3dkmdt.h` header file:
```
/////////////////////////////////////////////////////////////////////////////////////////////////////////////
// Purpose: Video output technology descriptor type.
//
// Remarks: Video output technology is used to determine the hard-coded list of video modes supported
//          by the monitor, when monitor descriptor is not available.
//
typedef enum _D3DKMDT_VIDEO_OUTPUT_TECHNOLOGY
{
    D3DKMDT_VOT_UNINITIALIZED        = -2,
    D3DKMDT_VOT_OTHER                = -1,
    D3DKMDT_VOT_HD15                 =  0,
    D3DKMDT_VOT_SVIDEO               =  1,
    D3DKMDT_VOT_COMPOSITE_VIDEO      =  2,
    D3DKMDT_VOT_COMPONENT_VIDEO      =  3,
    D3DKMDT_VOT_DVI                  =  4,
    D3DKMDT_VOT_HDMI                 =  5,
    D3DKMDT_VOT_LVDS                 =  6,
    D3DKMDT_VOT_D_JPN                =  8,
    D3DKMDT_VOT_SDI                  =  9,
    D3DKMDT_VOT_DISPLAYPORT_EXTERNAL = 10,
    D3DKMDT_VOT_DISPLAYPORT_EMBEDDED = 11,
    D3DKMDT_VOT_UDI_EXTERNAL         = 12,
    D3DKMDT_VOT_UDI_EMBEDDED         = 13,
    D3DKMDT_VOT_SDTVDONGLE           = 14,
#if (DXGKDDI_INTERFACE_VERSION >= DXGKDDI_INTERFACE_VERSION_WDDM1_3_M1)
    D3DKMDT_VOT_MIRACAST             = 15,
#if (DXGKDDI_INTERFACE_VERSION >= DXGKDDI_INTERFACE_VERSION_WDDM2_1)
    D3DKMDT_VOT_INDIRECT_WIRED       = 16,
#endif // (DXGKDDI_INTERFACE_VERSION >= DXGKDDI_INTERFACE_VERSION_WDDM2_1)
#endif // (DXGKDDI_INTERFACE_VERSION >= DXGKDDI_INTERFACE_VERSION_WDDM1_3_M1)
    D3DKMDT_VOT_INTERNAL             = 0x80000000,

    // Remove when DDI is unlocked.
    D3DKMDT_VOT_SVIDEO_4PIN          = D3DKMDT_VOT_SVIDEO,
    D3DKMDT_VOT_SVIDEO_7PIN          = D3DKMDT_VOT_SVIDEO,
    D3DKMDT_VOT_RF                   = D3DKMDT_VOT_COMPOSITE_VIDEO,
    D3DKMDT_VOT_RCA_3COMPONENT       = D3DKMDT_VOT_COMPONENT_VIDEO,
    D3DKMDT_VOT_BNC                  = D3DKMDT_VOT_COMPONENT_VIDEO,
}
D3DKMDT_VIDEO_OUTPUT_TECHNOLOGY;
```

It also does not clarify whether certain values can be OR-ed together when returned by the OS.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0400a08f-d9fe-de93-903c-be9a50797661
* Version Independent ID: 5f1b2d05-2217-40f5-73d3-effee4ef1c0e
* Content: [_D3DKMDT_VIDEO_OUTPUT_TECHNOLOGY (d3dkmdt.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmdt/ne-d3dkmdt-_d3dkmdt_video_output_technology)
* Content Source: [wdk-ddi-src/content/d3dkmdt/ne-d3dkmdt-_d3dkmdt_video_output_technology.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmdt/ne-d3dkmdt-_d3dkmdt_video_output_technology.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2022-07-29:
Hi @levicki - thanks for the issue. I've updated the page with the numerical values for each enum value. The only enum value that might make sense to bitwise OR with certain values is "D3DKMDT_VOT_INTERNAL = 0x80000000", but its description already states that it is not a bitfield value. I did find that the D3DKMDT_VIDEO_OUTPUT_TECHNOLOGY was incorrectly documented as a "bitwise OR" on the [DXGKMDT_OPM_STANDARD_INFORMATION ](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmdt/ns-d3dkmdt-_dxgkmdt_opm_standard_information) page, so fixed that. Appreciate you bringing this to my attention!

closed 2022-07-29 by lorihollasch

levicki, 2022-07-30:
Hi @lorihollasch You are welcome.

In the meantime, I have noticed through testing that `D3DKMDT_VOT_INTERNAL` is returned for laptop internal displays by the operating system (at least through `root\wmi` query `WmiMonitorConnectionParams`) as a standalone value -- i.e. it is never combined with the output type such as `D3DKMDT_VOT_LVDS` so that confirms it is meant to be used as a standalone value.

From this issue, a code analyzer rule could be devised -- do not mix bitfield and sequential enum values to avoid confusion unless they are meant to be used with bitwise OR to avoid the need to explicitly document whether it is possible or not.