MicrosoftDocs/windows-driver-docs-ddi #1338: ExAllocateFromNPagedLookasideList is not exported on Windows 10

fwang2002, 2022-08-02:
Why there are two entries for ExAllocateFromNPagedLookasideList in the document? The same declaration, but with different minimum supported client.

I try to work with [minispy driver sample](https://github.com/microsoft/Windows-driver-samples/tree/main/filesys/miniFilter/minispy) on Windows 10. It can be compiled, but can't be installed by `net start`.  See the [issue](https://github.com/microsoft/Windows-driver-samples/issues/761)
The Dependency Walker shows that there are missing exports of ExAllocateFromNPagedLookasideList and ExFreeToNPagedLookasideList in ntoskrnl.exe.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: baecb7dc-eeca-3831-7b21-d7eba5cf5fe9
* Version Independent ID: 4859d305-4818-c0e5-ac18-18cd26ffdb6c
* Content: [ExAllocateFromNPagedLookasideList function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatefromnpagedlookasidelist)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exallocatefromnpagedlookasidelist.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exallocatefromnpagedlookasidelist.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-08-12:
Thanks for the feedback @fwang2002 and apologies for the delay.  I just pushed a fix for the problem with the two documentation pages for ExAllocateFromPagedLookasideList that you describe.  In ~30 min or so, there should only be one doc page for this function.

fwang2002, 2022-08-15:
@tedhudek ExAllocateFrom... and ExFreeTo... still some function has duplicate entries.

And what I care about is: the ExAllocateFrom... and ExFreeTo... functions are not exported by ntoskrnl.exe on my windows 10, which does not consist with the document.

tedhudek, 2022-08-18:
Thanks - reopening to track removal of duplicate doc pages for these APIs:

- ExAllocateFromLookasideListEx
- ExFreeToLookasideListEx
- ExFreeToNPagedLookasideList

tedhudek, 2022-08-18:
Pushed fix, these should disappear from the live site within ~30 minutes

OZ1, 2023-03-22:
> @tedhudek ExAllocateFrom... and ExFreeTo... still some function has duplicate entries.
> 
> And what I care about is: the ExAllocateFrom... and ExFreeTo... functions are not exported by ntoskrnl.exe on my windows 10, which does not consist with the document. And because of the missing exports, I can't install official driver demo of minispy.

What about this issue? It is still present and should be addressed somehow.

tedhudek, 2023-03-22:
Hi @OZ1 and @fwang2002, I just noticed https://github.com/microsoft/Windows-driver-samples/issues/806#issuecomment-1381562218 discusses this problem.  Can you try the workaround there and let me know if it solves the problem?  If it does, I can add a note to those documentation pages.  If not, I will investigate further :)

(re-opening this issue while we discuss)

OZ1, 2023-03-23:
> 

Setting NTDDI_VERSION worked for me

closed 2023-03-24 by tedhudek