MicrosoftDocs/windows-driver-docs-ddi #1341: Please document `ZwQueryVirtualMemory(MemoryWorkingSetExInformation)`

gabriellandau, 2022-08-16:
Hello again.  Thank you for [documenting `FsRtlKernelFsControlFile`](https://github.com/MicrosoftDocs/windows-driver-docs/issues/3203).  We are already putting it to good use :)

I just [learned](https://windows-internals.com/understanding-a-new-mitigation-module-tampering-protection/) of a feature that's existed since approximately RS3/1709: `MEMORY_WORKING_SET_EX_BLOCK.SharedOriginal`.  Would you please document `MEMORY_INFORMATION_CLASS.MemoryWorkingSetExInformation`, which SystemInformer [claims](https://github.com/winsiderss/systeminformer/blob/42b4f1724c4c06487dc99825b9d85f74cc4f0cee/phnt/include/ntmmapi.h#L89) has been around since Vista, and the `SharedOriginal` bit from its corresponding structure, `_MEMORY_WORKING_SET_EX_BLOCK`, approximated [here](https://github.com/winsiderss/systeminformer/blob/42b4f1724c4c06487dc99825b9d85f74cc4f0cee/phnt/include/ntmmapi.h#L179-L219)?

The `SharedOriginal` bit will enable Antimalware products to efficiently identify executable image sections in memory that have been tampered, without the false positives & false negatives that can result from heuristics based on the `Shared` and `ShareCount` fields.  In fact, it is used by `ntdll!LdrpCheckPagesForTampering`.  With this information, Antimalware products will need to [scan](https://github.com/elastic/protections-artifacts) fewer loaded images, resulting in reduced CPU and I/O for our mutual customers.

If you're willing, could you also please document the same bit in [`PSAPI_WORKING_SET_EX_BLOCK`](https://docs.microsoft.com/en-us/windows/win32/api/psapi/ns-psapi-psapi_working_set_ex_block), indicating the version of Windows when it is first available?  We can test for support with [`IsWindowsVersionOrGreater`](https://docs.microsoft.com/en-us/windows/win32/api/versionhelpers/nf-versionhelpers-iswindowsversionorgreater).

lorihollasch, 2022-08-30:
Hi again @gabriellandau, glad you are enjoying FsRtlKernelFsControlFile docs. :) Regarding your other requests, we can only document entities (funcs, structs, enums, etc) found in publicly released header files:

- The released version of MEMORY_INFORMATION_CLASS (ntifs.h) contains MemoryBasicInformation only, so the docs are complete and correct as is.
- _MEMORY_WORKING_SET_EX_BLOCK is not in a released header file
- PSAPI_WORKING_SET_EX_BLOCK is Win32 (SDK) documentation, so beyond my purview, sorry! I was going to say open an issue through that page, but I don't see a way to do so, hmmm. Which bit specifically are you wanting documented in PSAPI_WORKING_SET_EX_BLOCK? I will pass your request along to the SDK team.

Sorry I can't be more helpful! You can direct your observations/requests RE Antimalware optimizations to the Filter team (perhaps try the MVI Yammer group).

gabriellandau, 2022-08-30:
> we can only document entities (funcs, structs, enums, etc) found in publicly released header files

Thank you for the clarification.  That's unfortunate.

> Which bit specifically are you wanting documented in PSAPI_WORKING_SET_EX_BLOCK? I will pass your request along to the SDK team.

It's the same `SharedOriginal` bit, but since it's not in a header, the same constraint will apply, so there's no need to pass the request along.  Thank you though.

> Sorry I can't be more helpful! You can direct your observations/requests RE Antimalware optimizations to the Filter team (perhaps try the MVI Yammer group).

You've been very helpful.  Thanks again!

closed 2022-08-30 by gabriellandau