MicrosoftDocs/windows-driver-docs-ddi #1345: The reply buffer size documentation is wrong

sri-mabelsoftware, 2022-08-22:
The documentation for FltSendMessage calls for the ReplyLength to be sizeof(FILTER_REPLY_HEADER) + sizeof(MY_STRUCT)
See below.
Therefore, it is recommended that you call [FilterReplyMessage](https://docs.microsoft.com/en-us/windows/win32/api/fltuser/nf-fltuser-filterreplymessage) and FltSendMessage (leveraging the above example) by setting dwReplyBufferSize and ReplyLength both to sizeof(FILTER_REPLY_HEADER) + sizeof(MY_STRUCT) instead of sizeof(REPLY_STRUCT).

But the scanner sample sets it to sizeof(MY_STRUCT)
Refer 
https://github.com/microsoft/Windows-driver-samples/blob/main/filesys/miniFilter/scanner/filter/scanner.c#L1327

I think doc is wrong.
Sri
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f6f940ae-4ddf-0e73-6607-32522f6ac5b7
* Version Independent ID: 20dfb40f-9d20-2d7b-9d81-4423eb0973de
* Content: [FltSendMessage function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltsendmessage)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsendmessage.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsendmessage.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2022-08-31:
Thanks for the issue, @sri-mabelsoftware. The docs actually are correct as written. Note that setting **dwReplyBufferSize** and **ReplyLength** to `sizeof(FILTER_REPLY_HEADER) + sizeof(MY_STRUCT) instead of sizeof(REPLY_STRUCT)` is the *recommendation* for robustness (and not a requirement).

closed 2022-08-31 by lorihollasch

sri-mabelsoftware, 2022-08-31:
You missed my noting - the size needs to be neither the sizeof(FILTER_REPLY_HEADER) + sizeof(MY_STRUCT) nor the sizeof(REPLY_STRUCT) - these two sizes could be obviously different and cause robustness issue. It should just be sizeof(MY_STRUCT) - i.e. dwReplyBufferSize does NOT need to include the header size of sizeof(FILTER_REPLY_HEADER) as per scanner sample.