MicrosoftDocs/windows-driver-docs-ddi #1352: IoVerifyVolume appears to leak VPB

billziss-gh, 2022-09-19:
It appears that `IoVerifyVolume` can leak the VPB of the passed `DeviceObject` in these 2 cases:

- The device has not been mounted before. (Confirmed on multiple versions of the OS: Client 10-2004, Server 2012 R2, Server 2016, Server 2019.)
- The device has been mounted before but the verification (`IRP_MN_VERIFY_VOLUME`) fails. (Not confirmed, but very likely.)

The problem appears to be that in either of these two cases the system will attempt to mount a file system on the passed `DeviceObject`. Prior to doing so it calls `IopCreateVpb`, which appears to blindly overwrite the `DeviceObject->Vpb` field. Thus the memory backing the original `DeviceObject->Vpb` is leaked.

In my file system driver I have confirmed this using the following approaches:

- By testing this [workaround](https://github.com/winfsp/winfsp/blob/5672a63409fb9b3584fa5ca461d50f067454fbe5/src/sys/volume.c#L401-L421) where I explicitly deallocate the old VPB after `IoVerifyVolume` returns. No ill effects (such as double free) are observed when doing so. Here is an example [appveyor run](https://ci.appveyor.com/project/billziss-gh/winfsp/builds/44806577) where this workaround is tested on multiple OS'es with the verifier on.

- By explicitly allocating a new VPB in my driver, setting it as the `DeviceObject->Vpb` prior to calling `IoVerifyVolume` and after completing my test observing that `!verifier 3` reports the leaked VPB memory.

My suggestion would be that the `IoVerifyVolume` documentation should be updated to mention this problem and suggest an appropriate workaround. There is also the possibility that I am simply misusing `IoVerifyVolume` and if that is the case please let me know.

lorihollasch, 2022-10-07:
Hey @billziss-gh, I've shared the info you provided with the dev team. They are super busy but said they'd look into it when they have time. I'll leave this issue open for now - hope to have more info for you soon. You could try posting your findings on an [OSR forum](https://www.osr.com/developer-community/) - let me know if you learn anything if you do so.

billziss-gh, 2022-10-07:
@lorihollasch thank you for your answer.

I have already posted a question in OSR, but have not received any response:

https://community.osr.com/discussion/293614/ioverifyvolume-and-vpb-leak

In the mean time I am no longer using `IoVerifyVolume` in my File System Driver until this issue is resolved.

lorihollasch, 2022-11-10:
@billziss-gh, apologies but the dev team still hasn't been able to find time to look at this issue. I took a quick look at your code - can you confirm that the DEVICE_OBJECT type is FILE_DEVICE_DISK for the IoVerifyVolume call? (I saw a few different FILE_DEVICE_XXX values)

lorihollasch, 2023-01-20:
Thanks for your patience, @billziss-gh. The feature team found some time and determined that this is "user error". I've updated [IoVerifyVolume](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ioverifyvolume) and [IoSetDeviceToVerify ](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-iosetdevicetoverify) with additional information and links to the FastFAT sample that show how to call these DDIs.

closed 2023-01-20 by lorihollasch

billziss-gh, 2023-01-20:
@lorihollasch thank you and the dev team for looking into this.