MicrosoftDocs/windows-driver-docs-ddi #1359: Need to document  FSCTL_QUERY_ALLOCATED_RANGES , including its restrictions

rodwiddowson, 2022-09-28:
Hey Lori,
I went looking for the page for  `FSCTL_QUERY_ALLOCATED_RANGES` because I wanted to make a change to it but it's not there.

I wouldn't normally advocate adding the documentation for things like this but for filter driver writers there is an important restriction which I feel needs documented.

It appears that the function `nt!FsRtlQueryCachedVdl` issues this FSCTL but looks at the returned status in terms of
```
if (!NT_ERROR(status)) {
   ProcessBuffer()
}
```
Rather than the more usual 
```
if (NT_SUCCESS(status) || (status = STATUS_BUFFER_OVERFLOW)) {
   ProcessBuffer()
}
```

(It also doesn't look at the returned length in the iosb, but that's another story)

The net result is that if any filter intercepts this call, calls some other function which fails with a WARNING (say `STATUS_DEVICE_OFF_LINE`) and then returns _that_ status up then  `nt!FsRtlQueryCachedVdl`  will party over the (non) returned buffer and bring the machine down which an access violation.

We can argue the toss about whether this is polite behavior, but the bottom line is that there are a great deal of deployed machines out there with this code in place, and a recent defender update appears to have become quite fond of that api.  So my belief is that this FSCTL needs documented, together with an explicit warning about "if you intercept this never return a WARNING status back except `STATUS_BUFFER_OVERFLOW` "

I can help write this if it would help.

Thanks, 

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: ce8fc339-4340-4f3e-c153-ce55aad1d007
* Version Independent ID: e8adb1a3-9bfb-912a-d1e3-30e8f623b5e3
* Content: [Ntifs.h header - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/)
* Content Source: [wdk-ddi-src/content/ntifs/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/index.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2022-09-28:
(in case someone wonders, this FSCTL does appear in the `10.0.22000.0` version of ntifs.  It also appears to have some user-mode-side documentation which I haven't perused too deeply

lorihollasch, 2022-10-19:
Hi @rodwiddowson - I've documented both [FSCTL_QUERY_ALLOCATED_RANGES](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ni-ntifs-fsctl_query_allocated_ranges) and [FSCTL_QUERY_FILE_REGIONS](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ni-ntifs-fsctl_query_file_regions), the latter FSCTL being the one with the issue you identify above. I shared your discovery with the dev team who should fix it as they are able. In the interim, I added a note in the Return Values section about checking IoStatus.Information before touching OutputBuffer as the recommended workaround. Thanks as always for your time and care in improving the docs and our products!

closed 2022-10-19 by lorihollasch

rodwiddowson, 2022-10-21:
That wording is a masterpiece.  Caps off to  @lorihollasch !