MicrosoftDocs/windows-driver-docs-ddi #1366: Needs IRQL environment definition

ivanrouzanov, 2022-10-06:
d3dkmthk.h defines PDXGK_INITIAL_COMPONENT_STATE as DISPATCH_LEVEL IRQL callback:

typedef
    _IRQL_requires_(DISPATCH_LEVEL)
VOID
(*PDXGK_INITIAL_COMPONENT_STATE) (
    PVOID GraphicsDeviceHandle,
    PVOID PrivateHandle,
    ULONG ComponentIndex,
    BOOLEAN IsBlockingType,
    UINT InitialFState,
    GUID ComponentGuid,
    UINT PowerComponentMappingFlag
);

What this means is that IRQL will be raised to DISPATCH_LEVEL when the callback is called (DXGGLOBAL::RegisterSharedPowerComponent acquires a spinlock). So implementers of the PDXGK_INITIAL_COMPONENT_STATE callback has to make sure no pageable code or data is accessed, therefore it is important to document this requirement here.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: b7fc4685-72f7-3422-0b2e-601e39047050
* Version Independent ID: 26adbac7-a725-94a1-379c-5c7e3a538d21
* Content: [PDXGK_INITIAL_COMPONENT_STATE (d3dkmthk.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmthk/nc-d3dkmthk-pdxgk_initial_component_state)
* Content Source: [wdk-ddi-src/content/d3dkmthk/nc-d3dkmthk-pdxgk_initial_component_state.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmthk/nc-d3dkmthk-pdxgk_initial_component_state.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2022-10-07:
Thanks for filing this issue, @ivanrouzanov. Driver writers are [expected to understand Windows OS fundamentals](https://learn.microsoft.com/en-us/windows-hardware/drivers/display/roadmap-for-developing-drivers-for-the-windows-vista-display-driver-mo), so we don't (can't) spell out all requirements on every ref page. I did add the IRQL level to the table at the bottom of the page, and also added a link within the Remarks section to a page that discusses pageable code/data since you asked - we appreciate your desire to improve the docs!

closed 2022-10-07 by lorihollasch

ivanrouzanov, 2022-10-08:
Thank you Lori, I really appreciate it! I do agree that driver writers need to understand Windows OS requirements, but they often don't. The reason I asked for this is because I had to debug real bug in an audio driver which was calling KeWaitForSingleObject from this callback leading to a Bugcheck. Thank you!