MicrosoftDocs/windows-driver-docs-ddi #1374: KeGetCurrentIrql library is inaccurate

snowkoan, 2022-10-26:
[Enter feedback here]

Library | Hal.lib -> this is only true for x86 targets
On x64 targets pre-Win10, the function is defined inline
On x64 targets Win10+, the function is exported by ntoskrnl.lib

I found this the hard way when trying to build a Windows 7x64 driver with WDK 11. The details might be a little different than I have written, but there are definitely these three cases.

<br class="Apple-interchange-newline">
</body>
</html>

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: abb99b21-5b0a-9256-2cf7-29480371ffff
* Version Independent ID: 116198d1-1e57-94b1-6aab-cc5235233a13
* Content: [KeGetCurrentIrql function (wdm.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-kegetcurrentirql)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-kegetcurrentirql.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-kegetcurrentirql.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2022-11-01:
Thanks @snowkoan for the feedback, apologies for the delay in getting back to you.  Do you recall what you had to do to fix the problem?  One thing to note is that WDK 11 officially supports building drivers only for Win10+.  The matrix at https://learn.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#step-2-install-the-wdk shows the kits to use to build for older versions of Windows.

snowkoan, 2022-11-02:
Hello @tedhudek . Thanks for the update. I'm building with WDK 11 21H2, which does support Windows 7+.

In order to compile my Windows 7 x64 and Windows 8.1 x64 targeted drivers, I had to define NT_INLINE_GET_CURRENT_IRQL .

Here's the definition of KeGetCurrentIrql from wdm.h:

```
#if defined(NT_INLINE_GET_CURRENT_IRQL) || defined(_NTHALLIB_)

_IRQL_requires_max_(HIGH_LEVEL)
_IRQL_saves_
CFORCEINLINE
KIRQL
KeGetCurrentIrql (
    VOID
    )

/*++

Routine Description:

    This function return the current IRQL.

Arguments:

    None.

Return Value:

    The current IRQL is returned as the function value.

--*/

{

#if !defined(_ARM64EC_)

    return (KIRQL)ReadCR8();

#else

    return 0;  // ARM64EC_STUB

#endif // !defined(_ARM64EC_)

}

#else

_IRQL_requires_max_(HIGH_LEVEL)
_IRQL_saves_
NTKERNELAPI
KIRQL
KeGetCurrentIrql (
    VOID
    );

#endif
```

tedhudek, 2022-11-22:
Ok thanks. I updated the Library metadata to NtosKrnl.lib.  That update will appear on the live site shortly.

closed 2022-11-22 by tedhudek