MicrosoftDocs/windows-driver-docs-ddi #1377: WdfInterruptSetPolicy() not working?

msudhakargit, 2022-10-29:
[Enter feedback here]

Hi

I am following what this page suggests, i.e. use WdfInterruptSetPolicy() to have desired settings for all my (MSI-X) interrupts. 
Then  when FxInterrupt::FilterResourceRequirements() is executed, I observe below
1. m_Policy/m_Priority/m_Processors.Mask are the values I specified on my corresponding WdfInterrupt Object
2. But IoResourceDescriptor->Flags has bit CM_RESOURCE_INTERRUPT_POLICY_INCLUDED set and hence it does not go inside the if() block for my settings to take effect (IoResourceDescriptor->Flags value was 7)

My device policy setting in INF is below, i.e. I do not have any AssignmentSetOverride key specified
             HKR, "Interrupt Management\Affinity Policy", DevicePolicy, %RegTypeDW%, 4
Also I do not register any WDF_FDO_EVENT_CALLBACKS, since I just want to use the WdfInterruptSetPolicy() way.

Given above I am not sure how/where the CM_RESOURCE_INTERRUPT_POLICY_INCLUDED bit got set, which is making WdfInterruptSetPolicy() invoke by the driver void.

Looks like others have same issue as well.
https://community.osr.com/discussion/258486/problem-setting-interrupt-affinity

Are there any other specific driver/INF requirements for this to be working for the drivers?

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 8cacd3e7-de7a-d800-6a67-ce983d52e964
* Version Independent ID: 36389474-9d07-25cd-a51f-982c717fc3c3
* Content: [WdfInterruptSetPolicy function (wdfinterrupt.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfinterrupt/nf-wdfinterrupt-wdfinterruptsetpolicy)
* Content Source: [wdk-ddi-src/content/wdfinterrupt/nf-wdfinterrupt-wdfinterruptsetpolicy.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfinterrupt/nf-wdfinterrupt-wdfinterruptsetpolicy.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

msudhakargit, 2022-10-29:
Hi

In debugger when i change the IoResourceDescriptor->Flgas to 3 (i.e. remove flag CM_RESOURCE_INTERRUPT_POLICY_INCLUDED), everything works fine. So looks like somewhere that flag gets set erroneously even for WdfInterruptSetPolicy() paths, thereby negating the setting.

Thanks

tedhudek, 2022-11-01:
Based on https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/interrupt-affinity-and-priority, I think that if you set DevicePolicy to 4, then you also need to specify AssignmentSetOverride:

 > If DevicePolicy is 0x04 (IrqPolicySpecifiedProcessors), then this mask specifies a set of processors to assign the device's interrupts to.

We also recommend that you set your interrupt affinity either all in the INF file, or all in the driver code, but not a mix of the two.

msudhakargit, 2022-11-01:
I also tested with DevicePolicy = 5 with INF, I see the same issue.

msudhakargit, 2022-11-01:
O.k. I will also try with no InterruptAffinity Policy set in the INF at all.

msudhakargit, 2022-11-03:
With no Interrupt Affinity/Device Policy set, the API works after install. But if I disable/enable the driver or restart the system the issue happens again.

closed 2022-12-07 by tedhudek