MicrosoftDocs/windows-driver-docs-ddi #1380: FOUND error

skrandy, 2022-11-03:
[Enter feedback here]
NTSYSAPI NTSTATUS RtlUnicodeStringToAnsiString(
  [in, out] PANSI_STRING     DestinationString,
  [in]      PCUNICODE_STRING SourceString,
  [in]      BOOLEAN          AllocateDestinationString
);

the PCUNICODE_STRING should be PUNICODE_STRING.
if it's my mistake,I'm sorry

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 27ec1ad4-44fd-4c25-f5bf-ab9f108e30bc
* Version Independent ID: 92211920-dcc6-3247-c353-84fb436c96e0
* Content: [RtlUnicodeStringToAnsiString function (wdm.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtlunicodestringtoansistring#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlunicodestringtoansistring.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlunicodestringtoansistring.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

mhopkins-msft, 2022-11-12:
Hi @skrandy 

I see the declaration in the wdm.h header looks like this:

```
NTSYSAPI
NTSTATUS
NTAPI
RtlUnicodeStringToAnsiString(
    PANSI_STRING DestinationString,
    PCUNICODE_STRING SourceString,
    BOOLEAN AllocateDestinationString
    );
```

It looks like this ref page is correct.

Where did you see that the type should be `PUNICODE_STRING`?

Thanks,
 ~ *Mark*

skrandy, 2022-11-14:
OK，Sorry bro。
I‘m not see this one:
typedef UNICODE_STRING *PUNICODE_STRING;
typedef const UNICODE_STRING *PCUNICODE_STRING;

 it's my mistake,I'm sorry.

mhopkins-msft, 2022-11-14:
Hi @skrandy, 

No worries, thank you for reaching out to us. I appreciate hearing from you.

 ~ *Mark*

closed 2022-11-14 by mhopkins-msft