MicrosoftDocs/windows-driver-docs-ddi #1403: Filtering is not contextual

raffaeler, 2022-12-23:
I landed in this page, which is kernel-specific:
https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/_kernel/

I tried to search HTTP, because I am looking for the APIs exposed by HTTP.SYS driver so that my driver can use those APIs.
This search brings me to Office, Windows, dotnet, and whatever else is available on the learn website. It is impossible to find anything.
The goal of the search is not to find as many as possible results, the goal should be to find the closest to the **context** I started from.

tedhudek, 2023-01-04:
Hi Raffaele, I'm sorry I can't be of much help with your question.  I looked through the driver documentation, which is where you landed, and the only reference I saw to http.sys was on this page -- https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xfa---http-driver-corrupted -- which is definitely not what you want.

Can you give me any more context on what you're trying to do, or doc pages that are close to what you want?  (The latter might help me find someone who can help.)

tedhudek, 2023-01-04:
Also, this page has some info about it.  

https://learn.microsoft.com/en-us/troubleshoot/developer/webapps/iis/iisadmin-service-inetinfo/httpsys-registry-windows

raffaeler, 2023-01-04:
Hi @tedhudek the context is quite simple. I am writing a driver (therefore running in ring0 / kernel mode) and I would like to use the HTTP.SYS kernel mode API to manage the TLS handshake.
The last link you posted is about the registry configuration of HTTP.SYS for ring3 (applications / user mode) usage.

The main and general issue with `MicrosoftDocs` is that you can't search a subtree of the documentation. This is very limiting.
The second and more specific issue is that I cannot make a search that is directed to the ring0 mode (which is a totally different developer environment). Let's call this "documentation partition" where any other result coming from other "parititions" is certainly wrong.
Finally, the third issue, I don't know where (if any) can be found the HTTP.SYS ring0 API.

Thank you

tedhudek, 2023-01-04:
Hi @raffaeler,

Http.sys is not available for kernel mode callers.
However, schannel is available in kernel mode through SSPI to perform TLS:

* https://learn.microsoft.com/en-us/windows/win32/api/sspi/nf-sspi-sspiacquirecredentialshandleasyncw
* https://learn.microsoft.com/en-us/windows/win32/api/sspi/nf-sspi-sspiacceptsecuritycontextasync

If you need HTTP server support from http.sys then you'll need to have a UM component call http.sys through httpapi.dll.

Sorry I don't have any control over search.  One thing you can do is clone the doc repos that interest you and search them locally.  Use the browser's View Source to see metadata on any Learn page and you can find the repo location there, for example:

`<meta name="github_feedback_content_git_url" content="https://github.com/MicrosoftDocs/windows-driver-docs/...`

Hope this helps!

closed 2023-01-04 by tedhudek

raffaeler, 2023-01-05:
Thanks for your answer @tedhudek