MicrosoftDocs/windows-driver-docs-ddi #1417: `HidP_GetUsages` (mistakenly?) specifies `Report` as `[out]`

seritools, 2023-02-05:
The `Report` parameter in [`HidP_GetUsages`](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/758b904a9035ee00c0bea6a2ed34a78202bcf8ce/wdk-ddi-src/content/hidpi/nf-hidpi-hidp_getusages.md) seems to be mistakenly marked as an `[out]` parameter, when it seems like it should be an `[in]` parameter instead, just like all the other `HidP_Get*` functions that take a `Report`.

tedhudek, 2023-02-06:
Thanks for the report Dennis.

Assigning to Mark to investigate.  Relevant doc page is https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/hidpi/nf-hidpi-hidp_getusages.

mhopkins-msft, 2023-03-04:
Hi @seritools,

Thank you for reporting this apparent discrepancy. I checked the hidpi.h header file and verified that `Report` is actually an `[out]` parameter. Here is the function prototype from hidpi.h:

```cpp
NTSTATUS __stdcall
HidP_GetUsages (
   _In_ HIDP_REPORT_TYPE    ReportType,
   _In_ USAGE   UsagePage,
   _In_opt_ USHORT  LinkCollection,
   _Out_writes_to_(*UsageLength, *UsageLength)  PUSAGE UsageList,
   _Inout_    PULONG UsageLength,
   _In_ PHIDP_PREPARSED_DATA PreparsedData,
   _Out_writes_bytes_(ReportLength)    PCHAR Report,
   _In_ ULONG   ReportLength
   );
```

You're right that other `HidP_Get*` functions have `Report` declared as an `[in]` parameter. I will pass this information on to the engineering team for possible updating in the future.

Thanks for reaching out to us!
 ~ *Mark*

closed 2023-03-04 by mhopkins-msft

Lokathor, 2023-03-04:
Perhaps this issue should stay open still? The MSDN and header have definitely always been in agreement, but that was not the really the point of the issue. The issue is asking if the headers themselves are wrong about the function.

mhopkins-msft, 2023-03-04:
Hi @Lokathor,

I agree that this function is inconsistent with other `HidP_Get*` functions, but our job on the docs team is to accurately document the code.

The actual implementation of the function also has this for the `Report` parameter.

```cpp
_Out_writes_bytes_(ReportLength)    PCHAR Report,
```

Since the documentation is correct according to the implementation and the header file, please follow the link at the bottom of the page to submit feedback on **[This product](https://support.microsoft.com/windows/send-feedback-to-microsoft-with-the-feedback-hub-app-f59187f8-8739-22d6-ba93-f66612949332)** versus **This page**. There you can propose that the engineering team reconsider this parameter declaration.

Thanks again for reaching out to the docs team,
 ~ *Mark*

Lokathor, 2023-03-04:
If that's what the actual implementation says then that answers the question. Thanks!