MicrosoftDocs/windows-driver-docs-ddi #1418: Please document `SeCaptureSubjectContextEx`

gabriellandau, 2023-02-07:
Hello again,

Thanks for all your help so far.

## SeCaptureSubjectContextEx

`SeCaptureSubjectContextEx` is listed in `10.0.22621.0/km/ntifs.h` in the 22H2 EWDK.  Could you please document it to help ensure safe usage?

```c
#if (NTDDI_VERSION >= NTDDI_VISTA)
NTKERNELAPI
VOID
SeCaptureSubjectContextEx (
    _In_opt_ PETHREAD Thread,
    _In_ PEPROCESS Process,
    _Out_ PSECURITY_SUBJECT_CONTEXT SubjectContext
    );
#endif
```

## SeQuerySubjectContextToken

This one is lower priority for us.  Right above it in `10.0.22621.0/km/ntifs.h` is `SeQuerySubjectContextToken` - that should probably have a security warning to ensure the impersonation level is checked to be `>= SecurityImpersonation` before using `ClientToken` for any security-related decisions.  Otherwise, attacks like [this](https://googleprojectzero.blogspot.com/2015/02/a-tokens-tale_9.html) are possible.

```c
//++
//
//  PACCESS_TOKEN
//  SeQuerySubjectContextToken(
//      _In_ PSECURITY_SUBJECT_CONTEXT SubjectContext
//      );
//
//  Routine Description:
//
//      This routine returns the effective token from the subject context,
//      either the client token, if present, or the process token.
//
//  Arguments:
//
//      SubjectContext - Context to query
//
//  Return Value:
//
//      This routine returns the PACCESS_TOKEN for the effective token.
//      The pointer may be passed to SeQueryInformationToken.  This routine
//      does not affect the lock status of the token, i.e. the token is not
//      locked.  If the SubjectContext has been locked, the token remains locked,
//      if not, the token remains unlocked.
//
//--

#define SeQuerySubjectContextToken( SubjectContext ) \
        ( ARGUMENT_PRESENT( ((PSECURITY_SUBJECT_CONTEXT) SubjectContext)->ClientToken) ? \
            ((PSECURITY_SUBJECT_CONTEXT) SubjectContext)->ClientToken : \
            ((PSECURITY_SUBJECT_CONTEXT) SubjectContext)->PrimaryToken )
```

gabriellandau, 2023-02-08:
Thanks @tedhudek.  

Sorry I should have been clearer - [SeQuerySubjectContextToken](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-sequerysubjectcontexttoken) is documented, but it can be dangerous to use for security decisions because it can return either the impersonation or the primary token, without any information about which token was returned or the impersonation level.  Primary and impersonation tokens have different semantics.  The example given [here](https://googleprojectzero.blogspot.com/2015/02/a-tokens-tale_9.html) shows a vulnerability in a Microsoft driver, `ahcache.sys`, because it makes a security decision based on the user of an impersonation token.  

The docs for [SeQuerySubjectContextToken](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-sequerysubjectcontexttoken) state that you can use [`SeQueryInformationToken`](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-sequeryinformationtoken) on the returned token, which is true, but if you call `SeQueryInformationToken(TokenUser)` on a `SecurityIndentification` impersonation token decide whether to allow/deny an action, you're basically recreating the `ahcache.sys` vulnerability.  This vulnerability would not exist if the impersonation level were `SecurityImpersonation` or higher.

tedhudek, 2023-02-08:
Thanks Gabriel, that helps clarify.  Lori will take care of you from here on out.

lorihollasch, 2023-02-22:
@gabriellandau, I've published a page for [SeCaptureSubjectContextEx](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-secapturesubjectcontextex) and have added a note to [SeQuerySubjectContextToken](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-sequerysubjectcontexttoken) about the security issue. Your GitHub issues are a pleasure to work on - you're obviously very well informed. Thank you for taking the time to help improve our docs and security!

closed 2023-02-22 by lorihollasch

gabriellandau, 2023-02-22:
Thanks so much @lorihollasch @tedhudek !