MicrosoftDocs/windows-driver-docs-ddi #1428: The definition is error

lovelyidiot, 2023-03-10:
[Enter feedback here]
Based on the definition of an existing callback, DrawIndex will output black screen.
"UINT IndexCount, UINT StartIndexLocation, INT BaseVertexLocation" is right, 

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 5c315319-bba6-5968-4b06-2bb2c9c76672
* Version Independent ID: f95a2522-6535-205d-e790-75d7410582e3
* Content: [PFND3D10DDI_DRAWINDEXED (d3d10umddi.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/d3d10umddi/nc-d3d10umddi-pfnd3d10ddi_drawindexed)
* Content Source: [wdk-ddi-src/content/d3d10umddi/nc-d3d10umddi-pfnd3d10ddi_drawindexed.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3d10umddi/nc-d3d10umddi-pfnd3d10ddi_drawindexed.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2023-03-13:
@lovelyidiot, what specifically do you think is incorrectly documented on this page? It sounds correct to me ...

lovelyidiot, 2023-03-15:
> @lovelyidiot, what specifically do you think is incorrectly documented on this page? It sounds correct to me ...

The callback PFND3D10DDI_DRAWINDEXED receives parameters "hDevice, IndexCount, BaseVertexLocation, StartIndexLocation" in runtime like ID3D11Context::DrawIndexed. But the description of the document is "hDevice, StartIndexLocation, BaseVertexLocation, IndexCount".

`
// this is right
VOID DrawIndexedCallback(D3D10DDI_HDEVICE hDevice, UINT IndexCount, UINT BaseVertexLocation, INT StartIndexLocation) {
  _D3D11Context->DrawIndexed(IndexCount, StartIndexLocation, BaseVertexLocation);
}
`