MicrosoftDocs/windows-driver-docs-ddi #1429: Weird rendering of correct markup

rodwiddowson, 2023-03-13:
Hey Lori, Ted, 

There is something odd about this page.

Looking at the source it definitely says

```
A pointer to a PVOID value. If (*<i>Destination</i>) = <i>Comperand</i>, then the routine sets (*<i>Destination</i>) to <i>Exchange</i>.
```

But it renders without the `*` in front of `<i>Destination</i>`.

Yet 

```
Specifies the PVOID value to set (*<i>Destination</i>) to.
```

renders correctly

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: ae06c717-d4e7-8614-9133-7a6e05f04d2c
* Version Independent ID: 45df979a-0f7e-42db-ee4a-b87a1ce671d5
* Content: [InterlockedCompareExchangePointer function (wdm.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-interlockedcompareexchangepointer?f1url=%3FappId%3DDev16IDEF1%26l%3DEN-US%26k%3Dk(WDM%252FInterlockedCompareExchangePointer)%3Bk(InterlockedCompareExchangePointer)%3Bk(DevLang-C%252B%252B)%3Bk(TargetOS-Windows)%26rd%3Dtrue)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-interlockedcompareexchangepointer.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-interlockedcompareexchangepointer.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2023-03-13:
Thanks Rod, that was a weird one indeed.  It turned out it affects double occurrences of `*<i>text</i>` on a single line.  I just updated all occurrences of that pattern (not just doubles) to use `**text*` instead.

tedhudek, 2023-03-13:
And the page you referenced now has the update --  https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-interlockedcompareexchangepointer

Thanks again Rod!

closed 2023-03-13 by tedhudek