MicrosoftDocs/windows-driver-docs-ddi #1431: incorrect major code - should be IRP_MJ_INTERNAL_DEVICE_CONTROL

PisarevSergey, 2023-03-17:
Actual Major Code is IRP_MJ_INTERNAL_DEVICE_CONTROL and not IRP_MJ_DEVICE_CONTROL

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 9a6df0ae-5f14-dac2-0da0-c23227cc3f03
* Version Independent ID: 0a82abd4-1c13-8068-752d-dee21a1d0027
* Content: [IOCTL_INTERNAL_BTHENUM_GET_DEVINFO (bthioctl.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/bthioctl/ni-bthioctl-ioctl_internal_bthenum_get_devinfo)
* Content Source: [wdk-ddi-src/content/bthioctl/ni-bthioctl-ioctl_internal_bthenum_get_devinfo.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/bthioctl/ni-bthioctl-ioctl_internal_bthenum_get_devinfo.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2023-03-17:
Hi @PisarevSergey

Thank you for contacting us about this. This value is determined directly from the source code by our build system that creates the final doc pages. I have no way to change this value manually. How are you determining that this value is incorrect? Perhaps we have a bug in our build tools that I should report.

Thank you,
 ~ *Mark*

PisarevSergey, 2023-03-17:
Hi Mark.

The name of ioctl itself contains a hint IOCTL_INTERNAL_BTHENUM_GET_DEVINFO - INTERNAL.

But the actual poof - if i send this ioctl to Bluetooth device as a regular ioctl i get BSOD.
However if i send this request as internal ioctl i'am getting correct values in output buffer.

mhopkins-msft, 2023-03-17:
Thank you for that information. I will do more research here and get resolved as soon as possible. Stay tuned...

mhopkins-msft, 2023-03-20:
Hi @PisarevSergey,

We've verified that this is an issue in our doc build system. We're currently working on a fix. 

Thank you for reporting this and for your research on the issue. I'm going to go ahead and close this GitHub issue. Look for a fix on these pages soon.

Thank you for helping to make the documentation better!
 ~ *Mark*

closed 2023-03-20 by mhopkins-msft

PisarevSergey, 2023-06-15:
Hi !
I believe I was using WdfIoTargetSendInternalIoctlSynchronously and WdfIoTargetSendIoctlSynchronously.

tedhudek, 2023-06-15:
Thanks Sergey.  We have a fix coming soon :)

tedhudek, 2023-06-15:
https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/bthioctl/ni-bthioctl-ioctl_bth_sdp_connect now has the correct major code.  Other INTERNAL pages coming soon.  Thanks again for bringing this to our attention Sergey.