MicrosoftDocs/windows-driver-docs-ddi #1433: Missing data types for all FWPS layer fields

foxx, 2023-03-22:
# The problem

All of the FWPS layer fields in `InFixedValues` are missing their associated data types.

For example,[ `FWPS_FIELDS_ALE_AUTH_CONNECT_V4`](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/a378a0e8ac6ec737ffd44cdede8534c55cf07651/wdk-ddi-src/content/fwpsk/ne-fwpsk-fwps_fields_ale_auth_connect_v4_.md?plain=1#L9) shows all of the associated fields, such as `FWPS_FIELD_ALE_AUTH_CONNECT_V4_ALE_APP_ID`, but it does not state what the data type is (e.g. u8, u16, bytearray etc).

There are some [examples](https://github.com/svn2github/peerblock/blob/master/src/pbfilter/filter_wfp.c#LL120C43-L120C90) scattered throughout GitHub, but these tend to only cover the most common fields. 

The only place where I should find a full listing of all the field types for every layer was using [WFP Explorer](https://github.com/zodiacon/WFPExplorer), a tool made by Pavel Yosifovich, co-author of the Windows Internals Books. After looking through the code of that repo, it seems that the field types are determined by using FWPS inspection functions during runtime, as opposed to something like an XML definitions file.

This can be problematic for both seasoned kernel developers and newcomers, as the only way to find the correct data type for a field is either through trial and error, example code or WFP Explorer. Although WFP Explorer will display field data types, it requires users to run a `.exe`, which isn't ideal.

# Proposed Solution

All field data types should be documented in their associated FWPS fields documentation page.

The value should be either a friendly string, as seen [here](https://github.com/zodiacon/WFPExplorer/blob/c18b27368406390239161db3c087721df1a3b4dc/WFPExplorer/StringHelper.cpp#L257) in WFP Explorer, or the exact name from [FWP_VALUE0](https://learn.microsoft.com/en-us/windows/win32/api/fwptypes/ns-fwptypes-fwp_value0), e.g. `byteArray16`

For example:

```
### -field FWPS_FIELD_ALE_AUTH_CONNECT_V4_IP_REMOTE_ADDRESS

The remote IP address.

Data Type: byteArray16
```

## Pull Request

If one of the maintainers can approve this then I'd be happy to submit a pull request with all the necessary changes (it's essentially an hours worth of data entry from WFP Explorer into the markdown files in this repo).

## Example from WFP Explorer

[image]

aviviano, 2023-03-29:
Thanks @foxx for opening this issue. I've passed this issue and your proposed solution to the development team and will let you know once I hear back.

aviviano, 2023-12-15:
Thanks again for pointing out this issue with FWPS layer fields, @foxx. We've opened an internal work item for these doc updates and will close this issue. We'll follow up with you when there is progress on the project.

closed 2023-12-15 by aviviano