MicrosoftDocs/windows-driver-docs-ddi #1437: WdfSpinLockAcquire/WdfSpinLockRelease documentation is missing, and incorrectly linked

peterwie, 2023-04-01:
MSDN search for WdfSpinLockAcquire & WdfSpinLockRelease only shows results about driver rules, not the API references.

Likewise the links under See Also to those functions point to the driver rules.  But it seems the API reference pages for acquire/release are missing.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 62528a70-6ac6-a297-2953-e5f63bdc4782
* Version Independent ID: cc6095ed-46b7-dc34-66ec-7fd79b49d057
* Content: [WdfSpinLockCreate function (wdfsync.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfsync/nf-wdfsync-wdfspinlockcreate)
* Content Source: [wdk-ddi-src/content/wdfsync/nf-wdfsync-wdfspinlockcreate.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfsync/nf-wdfsync-wdfspinlockcreate.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2023-04-12:
Great find, Peter.  Working on a fix...

tedhudek, 2023-04-30:
Thanks Peter, sorry for the delay.  These pages are now live:

- https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfsync/nf-wdfsync-wdfspinlockacquire
- https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfsync/nf-wdfsync-wdfspinlockrelease

closed 2023-04-30 by tedhudek