MicrosoftDocs/windows-driver-docs-ddi #1443: ExGetCurrentResourceThread  is orphaned

rodwiddowson, 2023-04-17:
Hey Lori, Ted,

This page (and 7 others) references `ExGetCurrentResourceThread()`

If you click on the link it throws you into the old docset and a quick grep through the repo shows only uses, not definitions (so I couldn't do a PR for you.

I think

     /Rod

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: fd9c7ce1-214e-fef0-9c03-03c29b2c4b2d
* Version Independent ID: f7d915e2-0be7-7437-9492-b97599da362a
* Content: [ExSetResourceOwnerPointerEx function (wdm.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exsetresourceownerpointerex)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exsetresourceownerpointerex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exsetresourceownerpointerex.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2023-04-29:
Hey Rod, sorry for the delay.  ExGetCurrentResourceThread doesn't generate a stub so it would be hard to doc.  We do have https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-psgetcurrentthread which is what it gets pound defined into:

`#define ExGetCurrentResourceThread() ((ULONG_PTR)PsGetCurrentThread())`

What change were you wanting to make to it?

rodwiddowson, 2023-04-30:
Hey Ted,
Thanks for getting back to me.  

The issue here is that this page reference twice to `ExGetCurrentResourceThread` and both times it is hyperlinked to the old doc set.

I suspect that at a minimum you could just pull the link from this (and all the similar pages - a quick grep finds 8 refereneces within `windows-driver-docs-ddi` and two in `windows-driver-docs`).

I'm guessing, however, that you have a standard form of words for this kind of situation which will allow you to link it to the real function.

Thanks

tedhudek, 2023-04-30:
Thanks for the feedback, Rod.  I added a bit to the Remarks of https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-psgetcurrentthread to give some coverage to ExGetCurrentResourceThread.  Then I updated the links to point to this page rather than the one on the previous-versions site.  Let me know if this looks reasonable to you.

Appreciate all your suggestions on the docs!

closed 2023-04-30 by tedhudek