MicrosoftDocs/windows-driver-docs-ddi #1444: Wrong information in Syntax field and outdated in LinkSpeed description

kdluzews, 2023-04-24:
The Description section mentions `DUMMYSTRUCTNAME`, which is also present in WDK headers, whereas in this article an anonymous structure is used instead.

The second thing I noticed is that table `LinkSpeed` is outdated. It only contains values for PCIe gen. 1 and 2. It lacks values for gen. 3, 4, 5 and 6.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: dc5e23f9-2295-d301-45a6-9e98584f79a8
* Version Independent ID: b94b7e73-1c47-3212-d0c9-b0aaa832d595
* Content: [_PCI_EXPRESS_LINK_STATUS_REGISTER (ntddk.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ns-ntddk-_pci_express_link_status_register)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_pci_express_link_status_register.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_pci_express_link_status_register.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2023-05-09:
Thanks for letting us know Karol.  I updated the syntax block on the doc page.  We are working on the other issue you reported.

tedhudek, 2023-05-20:
Thanks again Karol, we just published updates in #1454.  The updated docs are live on learn.microsoft.com, for example https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ns-ntddk-_pci_express_link_status_register.

We updated many pages as a result of your feedback.  Thanks again for caring about the docs!

closed 2023-05-20 by tedhudek