MicrosoftDocs/windows-driver-docs-ddi #1451: IoRegisterFsRegistrationChange deprecated

pagable, 2023-05-12:
This page should make it clear this function is deprecated and include a reference to the FltMgr documentation. 

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: bba8a474-eed8-348f-232d-157ab3ba149e
* Version Independent ID: 65f64487-6b78-800e-2425-61d59e37a9ae
* Content: [IoRegisterFsRegistrationChange function (ntifs.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ioregisterfsregistrationchange)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ioregisterfsregistrationchange.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ioregisterfsregistrationchange.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

craigbarkhouse, 2023-05-31:
Can you be more specific as to why you think IoRegisterFsRegistrationChange is deprecated?  It does not look deprecated to me.  There is a newer API IoRegisterFsRegistrationChangeMountAware that lets you pass a boolean in order to synchronize against mounts.  However IoRegisterFsRegistrationChange is still supported.  It is equivalent to IoRegisterFsRegistrationChangeMountAware passing FALSE.

pagable, 2023-05-31:
Calling this API will fail if the system is configured to block legacy file system filters. That effectively makes it deprecated for external use, which is fine because everyone should be writing minifilters anyway.

craigbarkhouse, 2023-05-31:
We should update the return values to mention that, thanks.

lorihollasch, 2023-06-06:
@pagable , thanks for drawing attention to the case of legacy filter blocking. I've updated the page for this situation. Many thanks to @craigbarkhouse for your help here!

closed 2023-06-06 by lorihollasch