MicrosoftDocs/windows-driver-docs-ddi #1460: Isn't ObjectId (and internal ID) available on REFS?

rodwiddowson, 2023-06-11:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a8ff02da-a99a-a848-79b5-01bccd1d3937
* Version Independent ID: 42baa81f-5dfe-c967-1d99-49c83da53b29
* Content: [FILE_OBJECTID_INFORMATION (ntifs.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_objectid_information)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_file_objectid_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_file_objectid_information.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

craigbarkhouse, 2023-06-13:
Object IDs are not supported on ReFS.

craigbarkhouse, 2023-06-13:
The most likely reason is due to ReFS's use of 128-bit File IDs, they could not find a clean way to distinguish File ID vs Object ID when processing an open by ID.

On NTFS, File IDs are 64-bit so they are easily distinguishable from Object IDs.  But note NTFS also simultaneously supports opening by 128-bit ID; if the upper 64 bits are all zero it's treated as an open by File ID, otherwise it's treated as an open by Object ID.

lorihollasch, 2023-06-13:
Thanks for the details, @craigbarkhouse! If @rodwiddowson thought it was important enough to ask, I figure it's important enough for everyone else to know. So, I added Craig's comments to the page. Thanks to you both!

closed 2023-06-13 by lorihollasch