MicrosoftDocs/windows-driver-docs-ddi #1473: Bad link in page

efraimN, 2023-08-08:
The PsGetCurrentProcess link is broken...
Does not links to the correct page. (IoGetCurrentProcess)

10x...

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 28a0c76c-c19b-a36d-e6ef-b4e7f0a15d63
* Version Independent ID: 66ce88d8-bd68-beb7-7b36-89cc30fced50
* Content: [IoGetCurrentProcess function (wdm.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iogetcurrentprocess)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iogetcurrentprocess.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iogetcurrentprocess.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

tedhudek, 2023-08-09:
Hi @efraimN thanks for the feedback.  I do not repro this problem.  On https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iogetcurrentprocess, there are 3 instances of PsGetCurrentProcess, and none of them are links (by design).  Were you maybe looking at a different page or link?

tedhudek, 2023-08-11:
Closing out for now, please reopen if still needed

closed 2023-08-11 by tedhudek