MicrosoftDocs/windows-driver-docs-ddi #1474: Please document `SeLocateProcessImageName`

gabriellandau, 2023-08-08:
Hello again!  `SeLocateProcessImageName` (described by CodeMachine [here](https://codemachine.com/articles/top_ten_kernel_apis.html)) provides kernel drivers with a way to query an EPROCESS's image's filename:

```
NTSTATUS
SeLocateProcessImageName(
    _Inout_ PEPROCESS Process,
    _Outptr_ PUNICODE_STRING *pImageFileName
    );
```

It's more performant than the following method of obtaining this information because it doesn't require any handle creation/destruction/references and can be done in a single function call.

1. `ObOpenObjectByPointer(pProcess, &hProcess)`
2. `ZwQueryInformationProcess(hProcess, ProcessImageFileName)` to get length
3. `ExAllocatePool`
4. `ZwQueryInformationProcess(hProcess, ProcessImageFileName)` to get the path
5. `ZwClose(hProcess)`

Since it's listed in ntifs.h in at least the 22000.0 & 22621.0 EWDKs, could you please document it for safe use?

lorihollasch, 2023-08-14:
Hi again, @gabriellandau! Thanks as always for your issues. This function's refpage is now live: https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-selocateprocessimagename

closed 2023-08-14 by lorihollasch

gabriellandau, 2023-08-14:
Looks great.  Thanks @lorihollasch!