MicrosoftDocs/windows-driver-docs-ddi #1496: 'Current process' needs some clarification

camptatopenmars, 2023-11-07:
[Threads have an owning process and then a process whose address space they are attached to, typically these are the same process, however a thread can be attached to an arbitrary process.  Some experimentation has shown that IoGetCurrentProcess() returns the EPROCESS that the current thread is attached to, not necessarily the threads owning process.  PsGetCurrentProcessId() however seems to return the owning process for the current thread even if the thread is attached to a different process at the time of the call.  As such EPROCES objects looked up using these two routines will not necessarily be the same EPROCESS objects.]

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: 28a0c76c-c19b-a36d-e6ef-b4e7f0a15d63
* Version Independent ID: 66ce88d8-bd68-beb7-7b36-89cc30fced50
* Content: [IoGetCurrentProcess function (wdm.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iogetcurrentprocess)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iogetcurrentprocess.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iogetcurrentprocess.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2024-01-08:
@camptatopenmars, thanks for your feedback on this article.

You may have success posting a question about the current process return value from IoGetCurrentProcess in the following forums, where other driver developers and Microsoft engineers can be found. 

[Microsoft Q&A: Windows Hardware development](https://docs.microsoft.com/answers/topics/windows-hardware.html)

[OSR Developer Community](https://community.osr.com/)

closed 2024-01-08 by barrygolden

camptatopenmars, 2024-01-08:
I dont have a question.  Just pointing out the docs are missing a detail
that may be important to a dev.