MicrosoftDocs/windows-driver-docs-ddi #1497: CurrentByteOffset information in FltReadFile/FltWriteFile

emilekwork, 2023-11-13:
The documentation for FltWriteFile and FltReadFile says: if the current file position is used, FltReadFile/FltReadFile updates the file object's CurrentByteOffset field by adding the number of bytes read when it completes the operation.

However specifying (or not specifying) the offset has no effect on whether FltReadFile/FltReadFile updates the file object's CurrentByteOffset field, because it’s the filesystem (e.g. NTFS) that updates the file offset, not FltReadFile/FltReadFile. Can this be confirmed and docs updated if required. Thanks.

lorihollasch, 2023-11-16:
@emilekwork, thanks for the issue. To answer your question: it's the call to FltRead/WriteFile that causes the update, regardless of which function did so under the covers (an API implementation is all the code that executes in response to calling it). That said, a review of the code found the description of ByteOffset to have some inaccuracies and missing key information. I've updated that parameter for FltReadFile(Ex) and FltWriteFile(Ex). I appreciate you drawing attention to this!

closed 2023-11-16 by lorihollasch

emilekwork, 2023-11-17:
Thank you for the update.