MicrosoftDocs/windows-driver-docs-ddi #1498: Handle to a Process ID?

poweradminllc, 2023-11-13:
This is confusing.  Typically a Process ID is an integer (DWORD).  So a HANDLE to a Process ID is.... the actual Process ID?  Or the handle to the process (and not the PID?).  The discussion about how to get the PEPROCESS is probably trying to be helpful, telling about a way to get it if it hadn't just been passed in.  What would be more helpful is knowing what functions can be used with the PEPROCESS.

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: da277b56-3966-a7d0-cd9d-5839a1bd1c62
* Version Independent ID: 94b90c09-8827-243d-e58e-4f8ef7dc2405
* Content: [PCREATE_PROCESS_NOTIFY_ROUTINE_EX (ntddk.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nc-ntddk-pcreate_process_notify_routine_ex)
* Content Source: [wdk-ddi-src/content/ntddk/nc-ntddk-pcreate_process_notify_routine_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nc-ntddk-pcreate_process_notify_routine_ex.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2024-01-08:
@poweradminllc, thanks for your feedback on this article.

You may have success posting a question about what functions can be used with the PEPROCESS in the following forums, where other driver developers and Microsoft engineers can be found. 

[Microsoft Q&A: Windows Hardware development](https://docs.microsoft.com/answers/topics/windows-hardware.html)

[OSR Developer Community](https://community.osr.com/)

closed 2024-01-08 by barrygolden

poweradminllc, 2024-01-08:
## Reply ABOVE THIS LINE to add a note to this request ##

Thank you.  I can find more about the PEPROCESS, but the documentation still seems like it is incorrect.  I can't think of any case where you get a HANDLE to an ID.  This must be some sort of typo.

Doug
Power Admin LLC

Email: ***@***.***
Web: https://www.poweradmin.com/<https://www.poweradmin.com/?ref=helpspot>
Support: http://support.poweradmin.com/<http://support.poweradmin.com/?ref=helpspot>
Phone: +1 800.401.2339
Fax: +1 866.266.8330

If you love our products or service, please help us get the word out!
A simple Tweet, forum post or email to a buddy helps!
Thank you!

Barry Golden - ***@***.*** - Jan 8 2024, 12:03 PM

@poweradminllc<https://github.com/poweradminllc>, thanks for your feedback on this article.

You may have success posting a question about what functions can be used with the PEPROCESS in the following forums, where other driver developers and Microsoft engineers can be found.

Microsoft Q&A: Windows Hardware development<https://docs.microsoft.com/answers/topics/windows-hardware.html>

OSR Developer Community<https://community.osr.com/>