MicrosoftDocs/windows-driver-docs-ddi #1502: "Query Method Flag Value" appear to assume that the target of the rename exists.

rodwiddowson, 2023-11-24:
This is a real nit but...

"The Query Method Flag Value" paragraph has a bad attack of cut and paste.  There is a pretty good chance that this stage that the destination file does **not** exist at this point so statements like "query the file system for the file name" need further explanation.

I'd guess that what is being queried is the target directory name (and/or the handle opened with SL_OPEN_TARGET) but this needs clarified (somehow).  I fear that this will mean careful editing of that entire section.  I'm sorry.

Incidentally as an architectural question it is entirely unclear to me why this call need to have so many parameters/  It could get away with the CBD, the NameOptions and the RetFileNameInformation

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: a380ed8e-cd20-fa24-230c-0b7da7ff4f3b
* Version Independent ID: 21149e9f-6321-4afa-50ff-233fa3cb804e
* Content: [FltGetDestinationFileNameInformation function (fltkernel.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltgetdestinationfilenameinformation)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetdestinationfilenameinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetdestinationfilenameinformation.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2023-12-05:
Another fantastic clarification request (not "nit"), @rodwiddowson! Page is updated to hopefully capture the "file likely doesn't exist" nuance. Really appreciate you!

closed 2023-12-05 by lorihollasch

lorihollasch, 2023-12-05:
I don't have an answer to your architectural question, @rodwiddowson, but it appears you are right that fewer params would have been just fine. But it is what it is. (Apologies for two messages per issue - can you tell it's late here!?)

cgallred, 2023-12-05:
@rodwiddowson about the architectural question: the reason for splitting the parameters out like it does is for flexibility. If a filter simply wanted to get the destination name as the caller specified it, then the CBD, NameOptions, and RetFileNameInformation would suffice. But, for example, if a filter wants to redirect a rename it could supply its own values for the arguments, different from what is in the CBD, and still be able to query the destination name.