MicrosoftDocs/windows-driver-docs-ddi #1505: error structure with _SRTL_ADVANCED_FCB_HEADER

Mas0nShi, 2023-12-01:
well, when i use about  FCB, i find the structure `_SRTL_ADVANCED_FCB_HEADER` in spec is different in `windbg`

In windbg

```
kd> dt _FSRTL_ADVANCED_FCB_HEADER
nt!_FSRTL_ADVANCED_FCB_HEADER
   +0x000 NodeTypeCode     : Int2B
   +0x002 NodeByteSize     : Int2B
   +0x004 Flags            : UChar
   +0x005 IsFastIoPossible : UChar
   +0x006 Flags2           : UChar
   +0x007 Reserved         : Pos 0, 4 Bits
   +0x007 Version          : Pos 4, 4 Bits
   +0x008 Resource         : Ptr64 _ERESOURCE
   +0x010 PagingIoResource : Ptr64 _ERESOURCE
   +0x018 AllocationSize   : _LARGE_INTEGER
   +0x020 FileSize         : _LARGE_INTEGER
   +0x028 ValidDataLength  : _LARGE_INTEGER
   +0x030 FastMutex        : Ptr64 _FAST_MUTEX
   +0x038 FilterContexts   : _LIST_ENTRY
   +0x048 PushLock         : _EX_PUSH_LOCK
   +0x050 FileContextSupportPointer : Ptr64 Ptr64 Void
   +0x058 Oplock           : Ptr64 Void
   +0x058 ReservedForRemote : Ptr64 Void
   +0x060 AePushLock       : Ptr64 Void
   +0x068 BypassIoOpenCount : Uint4B
   +0x070 ReservedContext  : Ptr64 _FSRTL_PER_STREAM_CONTEXT
```

but in [spec](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header)

```
typedef struct _FSRTL_ADVANCED_FCB_HEADER {
  FSRTL_COMMON_FCB_HEADER DUMMYSTRUCTNAME;
  PFAST_MUTEX             FastMutex;
  LIST_ENTRY              FilterContexts;
  EX_PUSH_LOCK            PushLock;
  PVOID                   *FileContextSupportPointer;
  union {
    OPLOCK Oplock;
    PVOID  ReservedForRemote;
  };
  PVOID                   AePushLock;
  PVOID                   ReservedContext;       // +0x068 BypassIoOpenCount : Uint4B
  ULONG                   BypassIoOpenCount;     // +0x070 ReservedContext  : Ptr64 _FSRTL_PER_STREAM_CONTEXT
} FSRTL_ADVANCED_FCB_HEADER;

```

---
#### Document Details

⚠ *Do not edit this section. It is required for learn.microsoft.com ➟ GitHub issue linking.*

* ID: cb2d22bd-f5a3-890a-6703-1b2a1d7047d3
* Version Independent ID: cebba538-54f4-41e5-b7ec-6e0e724cfac5
* Content: [FSRTL_ADVANCED_FCB_HEADER (ntifs.h) - Windows drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md)
* Product: **windows-hardware**
* Technology: **wdk-api-reference**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

Mas0nShi, 2023-12-02:
@lorihollasch Thank you for your reply.
Sure, You are right. but, I think you misunderstood me...
I know about this, but the windbg show the last members of `FSRTL_ADVANCED_FCB_HEADER ` structure is

```
   +0x068 BypassIoOpenCount : Uint4B
   +0x070 ReservedContext  : Ptr64 _FSRTL_PER_STREAM_CONTEXT
```

but spec shows:

```
  PVOID                   ReservedContext;
  ULONG                   BypassIoOpenCount;
} FSRTL_ADVANCED_FCB_HEADER;
```

lorihollasch, 2024-01-09:
@Mas0nShi, ohhhhhhhh. I did miss the nuance that you were reporting, sorry. This is not a doc bug - FSRTL_ADVANCED_FCB_HEADER's syntax block on the refpage is correctly generated by our toolset based on that structure's definition in ntifs.h. Windbg appears to be doing something odd. I'll leave this issue

DOMARS, 2024-01-09:
Hello @Mas0nShi and @lorihollasch - I work with the debugger and can help you provide feedback. 
For filing issues, the WinDbg team requests that you check the behavior on the latest version of the new WinDbg. I think you can click on check updates to make sure WinDbg is up to date and then gather the WinDbg version information.

I will assume you are using symbols normally, off of the Microsoft public symbol server. Sometimes you can use !sym noisy to see the GUID of the PDB for the symbol if we need to gather that information. The symbols file is used to render the output shown.

I don't normally work in the IFS area, so not sure I'm following the order in the header. Starting with `AePushLock` it is greater then I assume the version of Windows you are using, and is included in your output.

Next up is  `PVOID ReservedContextLegacy` which I would think would also be included, but I don't see it in your output. The comments say that it was used was used in FCB_HEADER_V3 to hold a reserved context. But maybe not used anymore.

Then I see   `BypassIoOpenCount`.

Lastly, I see  `ReservedContext`  which it says is used for FSRTL_FCB_HEADER_V5.

So that order in header lines up with what you are seeing in the WinDbg.

But there are three init macros and many flags, so not really sure how this really works.

I see in this topic, that behavior may vary, per mini-filter. So this is all pretty complicated for a non IFS person, like myself.

https://learn.microsoft.com/en-us/windows-hardware/drivers/ifs/bypassio

I found this thread, where OSR expert Scott Noone provided some information on using !fileobj to locate the address of a specific FILE_OBJECT structure in memory. Then use that address to look at a specific instance to an FCB _HEADER file object - that might be something to consider trying.

 `dt nt!_FSRTL_ADVANCED_FCB_HEADER 0xffff800e5786e170`

https://community.osr.com/discussion/comment/305842#Comment_305842

kd> !fileobj 0xffffdd8e`36f30680

\Windows\System32\cmd.exe

Device Object: 0xffffdd8e309408a0   \Driver\volmgr
Vpb: 0xffffdd8e309d1260
Access: Read SharedRead SharedDelete 

Flags:  0x44442
    Synchronous IO
    Cache Supported
    Queue Irps to Thread
    Cleanup Complete
    Handle Created

FsContext: 0xffff800e5786e170   FsContext2: 0xffff800e5afae380
CurrentByteOffset: 0
Cache Data:
  Section Object Pointers: ffffdd8e3405a818
  Shared Cache Map: 00000000

1: kd> dt nt!_FSRTL_ADVANCED_FCB_HEADER 0xffff800e5786e170

```cpp
#if (NTDDI_VERSION >= NTDDI_WIN10_MN)
    //
    //  The following field is valid only if the Version field in the
    //  FSRTL_COMMON_FCB_HEADER is >= to FSRTL_FCB_HEADER_V3.  This field is
    //  present in Windows 10 Manganese and beyond and must be initialized by
    //  calling FsRtlSetupAdvancedHeaderEx2.
    //
    //  This field is used instead of PushLock to synchronize access to the
    //  list of stream contexts.  It enables a type of push lock that can
    //  automatically change from a normal pushlock to a cache-aware pushlock
    //  when the lock is subject to high cache contention due to a large number
    //  of concurrent shared acquirers.
    //

    PVOID AePushLock;

#elif (NTDDI_VERSION >= NTDDI_WINBLUE)

    //
    //  This field was used in FCB_HEADER_V3 to hold a reserved context.
    //

    PVOID ReservedContextLegacy;

#endif

#if (NTDDI_VERSION >= NTDDI_WIN10_CO)
    //
    //  The following field is valid only if the Version field in the
    //  FSRTL_COMMON_FCB_HEADER is >= to FSRTL_FCB_HEADER_V4.  This field is
    //  present in Windows 10 Cobalt and beyond.
    //
    //  This tracks how many handles are currently open with BypassIO enabled on
    //  this stream.
    //

    ULONG BypassIoOpenCount;

#endif

#if (NTDDI_VERSION >= NTDDI_WIN10_NI)

    //
    //  The following field is valid only if the Version field in the
    //  FSRTL_COMMON_FCB_HEADER is >= to FSRTL_FCB_HEADER_V5.
    //
    //  This field is used internally by the FSRTL to assist in context lookup.
    //

    struct _FSRTL_PER_STREAM_CONTEXT *ReservedContext;
#endif
```

closed 2024-01-09 by lorihollasch