MicrosoftDocs/windows-driver-docs-ddi #1507: Customer feedback - Why STATUS_PRIVILEGE_NOT_HELD

rodwiddowson, 2023-12-18:
### Type of issue

Missing information

### Feedback

Given that WU insists on using them I have to write a test program for transactions to test my drivers.  Ugh. Gulp.

My current program is returning STATUS_PRIVILEGE_NOT_HELD from its one call to this function about 4 times in 5 (no change to the program).  My suspicion is that the created transaction lives longer than the process and so and I need to wait for it to exit.  Or something.

Either way the error is frequent enough that STATUS_PRIVILEGE_NOT_HELD seems too important to relegate to the "oh yea and there are other errors" bucket.  (STATUS_INSUFFICIENT_RESOURCES sure).

Thanks

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ntcreatetransaction

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ntcreatetransaction.md

### Author

@barrygolden

rodwiddowson, 2023-12-18:
NM.  The code I inherited wasn't initializing the DesiredAccess Field

Close this unless you think its useful.  Have a great festive season

tedhudek, 2023-12-19:
Hi Rod, do you think we should add anything to the page in question?  I note that for the DesiredAccess parameter the page does currently say: "The DesiredAccess value cannot be zero."  Maybe that's enough?

rodwiddowson, 2023-12-20:
@tedhudek im not sure, but I’m honoured that you asked me.

There’s something oddly specific about the error (not like STATUS_INVALID_PARAMETER for instance), I suspect my broken test program was throwing in an odd bit in somewhere. I guess that if I were you I’d crack open the code and if anything popped out document it, otherwise just close the case and we’ll pretend it never happened.

rodwiddowson, 2023-12-31:
@tedhudek  I don't know whether this helps, but I got bored and started poking.

It turns out that if you specify a DesiredAccess of 0x01000000 to `NtCreateTransaction` is says `STATUS_PRIVILEGE_NOT_HELD`

AFAICS that desired access is `ACCESS_SYSTEM_SECURITY` so this isn't entirely surprising (I only learned that that was a valid setting for an ACCESS_MASK 5 minutes ago).

You might want to just close this case off a PEBKAC/IIHDDI

Happy new year

tedhudek, 2024-01-04:
@rodwiddowson thanks for the further research. In your original message, you said you got `STATUS_PRIVILEGE_NOT_HELD` 4 times out of 5.  Were you specifying **DesiredAccess** of 0x01000000 all 5 times?

I note in the Return value table on https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-ntcreatetransaction that it says you should get `STATUS_INVALID_PARAMETER` if **DesiredAccess** is zero, and `STATUS_ACCESS_DENIED` if the value of **DesiredAccess** is invalid, but it doesn't address any other special cases for that parameter.

I will poke around a bit internally.

And PEBKAC is quite entertaining when one searches it up, but Bing gives me the International Hip Dysplasia Institute when I look up IIHDDI, likely not what you meant? 😂

HNY!

rodwiddowson, 2024-01-04:
@tedhudek 
HNY!

(If It Hurts, Don’t Do It).

The 4 out of 5 was when my test code was passing an uninitialised variable in to the call so I’m guessing that 4 out of 5 time that bit was set - which is where the Problem Existing Between Keyboard And Chair bit came in.

tedhudek, 2024-01-08:
Hey Rod thanks for the exchange here.  We do have the boilerplate "The routine might return other NTSTATUS values" in the Return value section for that page so I'm going to let that cover this edge case.

We always appreciate your detailed feedback :)

closed 2024-01-08 by tedhudek