MicrosoftDocs/windows-driver-docs-ddi #1514: Customer feedback - Does this function throw like all other Mm/Cc functions

rodwiddowson, 2024-01-10:
### Type of issue

Missing information

### Feedback

@lorihollasch Thanks for putting this page up.

One of the more "architecturally interesting" quirks of NT is that the Memory Manager and its companion-in-arms the Cache Manager signal error conditions by throwing an exception.  So my go-to on a new project is to shim things like `CcSetFileSizes` inside a try/catch.

This function returns a status which might lead one to believe that it doesn't throw an exception but I would be super leery of expecting that (I have seen exceptions from Cc functions which I really thought couldn't).

Sooo if this function doesn't throw an exception it would be nice if this was made explicit (because it would be very much the exception to the norm).

I'll not open the whole can of worms about documenting all the Mm/Cc functions that throw - it could be argued that if you know enough about these functions to use them you expect them to throw - NT File system internals 101 (or maybe 201)

Again, thank you & your collegues for making these documents so very very useful

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ccsetfilesizesEx

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ccsetfilesizesex.md

### Author

@lorihollasch

lorihollasch, 2024-01-10:
Can you clarify what more you're looking for, @rodwiddowson? The Remarks section states that CcSetFileSizesEx might raise an exception and should thus be wrapped in a try-except (or try-finally).

rodwiddowson, 2024-01-11:
Ugh. Mea Maxima Culpa, I looked for and missed that Not sure how.

This of course raises the obvious question of what value is added by having a function return a (synchronous) failure in two different ways.  But those questions are waay above our pay grade.

closed 2024-01-11 by rodwiddowson