MicrosoftDocs/windows-driver-docs-ddi #1515: Customer feedback - This API does not signal all possible errors

rodwiddowson, 2024-01-12:
### Type of issue

Typo

### Feedback

@lorihollasch try not to hate me too much 😄 

An eagle-eyed colleague pointed out that the implementation of this in some OS versions is just a call to `CcSetFileSizesEx` which then  ignores any error status from it.

This argues either that this function is not reporting all errors or that the 'ex' version never reports errors via a status.  Or it could mean that the errors reported in the status from the 'ex' version are less important,

It seems like something should be done I just don't know what.

For myself I now just use the 'ex' version and either use the status it returns or the status it throws.  This seems to cover all eventualities - I would certainly be leery of using this function

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ccsetfilesizes

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ccsetfilesizes.md

### Author

@lorihollasch

lorihollasch, 2024-01-12:
LOL, no one could ever hate you, @rodwiddowson! I'll run your comments past the dev team ...

lorihollasch, 2024-02-03:
OK, @rodwiddowson, I've heard back from the cache manager team and once again your issue is a great call-out. I've updated the Ex page with some key info RE error codes and exception raising. I've also updated the non-Ex page to recommend that people use the Ex version instead. Thanks as always. Have a great weekend!

closed 2024-02-03 by lorihollasch