MicrosoftDocs/windows-driver-docs-ddi #1519: Customer feedback -

belinea123, 2024-01-16:
### Type of issue

Missing information

### Feedback

The documentation is very vague about when the EVT_WDF_INTERRUPT_ENABLE callback is called. It says:

> The framework calls the driver's EvtInterruptEnable callback function each time the device enters its working (D0) state.

More precisely, the EVT_WDF_INTERRUPT_ENABLE callback is called after EvtDeviceD0Entry has returned (source: "Developing drivers with the Windows Driver Foundation", Orwick, Smith).

EVT_WDF_INTERRUPT_DISABLE callback is called before the framework calls EvtDeviceD0Exit.

I had to consult the mentioned book to learn about the workflow of pnp and interrupts. The Microsoft Doc should be clearer here IMHO.

### Page URL

https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfinterrupt/nc-wdfinterrupt-evt_wdf_interrupt_enable

### Content source URL

https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfinterrupt/nc-wdfinterrupt-evt_wdf_interrupt_enable.md

### Author

@tedhudek

wm1, 2024-01-19:
At the end of the `EvtInterruptEnable` page, there is:
 > For more information about handling interrupts in framework-based drivers, see [Handling Hardware Interrupts](https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/handling-hardware-interrupts).

From there, the next page is [Enabling and Disabling Interrupts](https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/enabling-and-disabling-interrupts), which says the following:
> The framework calls the driver's `EvtInterruptEnable` and `EvtDeviceD0EntryPostInterruptsEnabled` callback each time the device enters its working (D0) state, after the framework has called the driver's `EvtDeviceD0Entry` callback.

Another useful page is [Power-Up Sequence](https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/power-up-sequence-for-a-function-or-filter-driver) and [Power-Down and Removal Sequence](https://learn.microsoft.com/en-us/windows-hardware/drivers/wdf/power-down-and-removal-sequence-for-a-function-or-filter-driver).

If you believe the document is still not clear, please let us know. Thanks,

belinea123, 2024-01-24:
Thanks for the info. No change is necessary.

closed 2024-01-24 by belinea123